micropython/micropython #3742: External Dependencies for unix port missing a little something

osamarais, 2018-04-27:
This is a somewhat silly issue, but the external dependencies for the unix port does not mention that python2 needs to be available. Since python2 does not come with new versions of fresh ubuntu installs, it can be maddening to amateurs like me :)

stinos, 2018-04-27:
What piece of functionality requires Python2? I only have Python3 and have no problems, but maybe you're trying something different?

osamarais, 2018-04-27:
I had a fresh ubuntu and was attempting tk get the unix port working. I followed the instructions till the last command 'make'
At that point I got an error. I installed python2 and then it worked just fine.

stinos, 2018-04-29:
Ah. What you really need is `python` on the commandline, doesn't matter whether it's 2 or 3, so likely your problem is a fresh Ubuntu doesn't have /usr/bin/python symlinked to python3.

Previous discussion here: https://github.com/micropython/micropython/issues/1616, related: https://github.com/micropython/micropython/pull/1621; @dhylands is that the a complete fix for the issue seen here?

I agree this should be made more clear in the README though, especially because it is crucial information.  Now it just says "You will also need bash, gcc, and Python (at least 2.7 or 3.3)." But that's above the 'external dependencies' line. And it is not even completely correct, shells other than bash probably work as well, idem for other compilers, and idem for the statement about requiring make.

Actually, after trying to objectively read the README, I found it to be a bit messy:
- some ports are listed under 'Major components', others under 'Additional components'. That is confusing, maybe it should just list ports/ under 'Major components' and say it contains all ports, eventually list them all but in the same place, and then clearly mention each port has a README with extra infoand build instructions.
- the Unix/STM32 paragraps seem too detailed to go on the front page, this info could just be offloaded to their respective README in the ports subdirectory.
- likewise for the 'External Dependencies' paragrah: it tries to be generic and for each port, but in reality it's mainly just more build instructions for the unix port so it's not relevant for a high percentage of users
- the unix port seems to be the only one without a README, that is also confusing, and likely is the case because it's now in the main README

@dpgeorge the README is the first thing new users see so I think it deserves some extra attention to make it good, I can make a PR adressing concerns above if you agree with them.

dhylands, 2018-04-29:
Using the update alternatives (the second comment in #1616
<https://github.com/micropython/micropython/issues/1616> seems like the
simplest thing to do until such time as something like #1621
<https://github.com/micropython/micropython/pull/1621> is applied to the
tree.

closed 2022-06-23 by dpgeorge