micropython/micropython #3744: ESP8266 network fixed IP not working

osamarais, 2018-04-27:
Attempting to set a fixed IP using the documentation:
https://docs.micropython.org/en/latest/esp8266/library/network.html

    wlan.ifconfig(config=('192.168.178.107', '255.255.255.0', '192.168.178.1', '8.8.8.8'))

does not work.

    wlan.ifconfig(('192.168.178.107', '255.255.255.0', '192.168.178.1', '8.8.8.8'))
only APPEARS to change the IP address. When checking the router IP table it does not change!

dpgeorge, 2018-05-01:
This feature (setting fixed IP address in STA mode) is definitely implemented (using `wifi_set_ip_info`).  Testing it, it works correctly for me: I can set a custom IP address and DNS address and the old one is no longer ping'able but the new address is.

> When checking the router IP table it does not change!

Maybe the router did not update?  Did you try pinging the device?

osamarais, 2018-05-02:
I rebooted the router and it did not change.
Instead of pinging I tried to connect using the webrepl, and couldn't connect with the new IP.
I'll try pinging to get a clearer picture.

osamarais, 2018-05-03:
Okay I played around with it a bit more.
After setting the static IP using ifconfig() I had to reconnect to the WiFi using my SSID and password.
With this I was able to connect with a static IP, get ping responses, and open up a WebREPL.

But I guess it would be nice to have the ESP8266 automatically do this when the IP is changed.

dpgeorge, 2018-05-04:
> After setting the static IP using ifconfig() I had to reconnect to the WiFi using my SSID and password.

I didn't need to reconnect to the AP, setting the static IP worked right away.  It could be an issue with the WiFi router.

closed 2018-05-15 by osamarais