micropython/micropython #3746: UART readline documentation improvement

nagylzs, 2018-04-30:
I could only try this on the ESP8266 port: the uart.readline() call does not always reads data until the first newline. When the read buffer becomes empty, then readline() returns (even if there was no newline received). The documentation is not clear about this. It only says that uart.readline() will "read a line ending in a newline character". That is not entirely true. It should tell this: "read characters until the first newline character or until the read buffer becomes empty. The readline() method is non-blocking."

peterhinch, 2018-05-01:
On the Pyboard **uart.readline()** blocks until either a complete line has been received or the timeout occurs. In the latter case it returns the current buffer contents (if any).

dpgeorge, 2018-05-02:
The UART class does have a configurable timeout (at least on pyboard and esp8266) and as @peterhinch mentions the readline (and other read methods) will return early if the timeout is reached.  And yes, this behaviour could be better described in the docs.

jason-neal, 2020-01-08:
This issue can probably be closed now

closed 2020-01-08 by dpgeorge