micropython/micropython #3747: Poller.poll() never gives POLLHUP or POLLERR

pushkarprasad007, 2018-04-30:
`			sock = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
			sock.setsockopt(socket.SOL_SOCKET, socket.SO_REUSEADDR, 1)
			sock.connect((sockaddr, 8000))
			# Set to non blocking
			sock.setblocking(0)
			# Get secret to generate the hash
			secret = get_iot_secret()
			# Find the hash of secret+ts
			hash_bytes = sha256(secret + str(epoch)).digest()
			# Send the keepalive packet now
			send_keep_alive(sock, epoch, hash_bytes)
			poller = select.poll()
			poller.register(sock, select.POLLERR | select.POLLIN | select.POLLHUP)
			# Poll for a sec
			while True:
				events = poller.poll(500)  # time in milliseconds
				print(events)
				if events:
					for fd, flag in events:
						print(flag)
						if flag & select.POLLIN:
							# Event is there, means data is there, read the data
							# 250 bytes should be more than enough since most packets
							# would be not more than 100 bytes
							data = sock.recv(250)
							# Now try to do processing of the data received
							process_server_packet(data, sock)
						if (flag & select.POLLHUP) or (flag & select.POLLERR):
							poller.unregister(s)
							break
				# Do periodic Tasks
				perform_cron_task(sock)
				# Sleep for a small while - 500 ms
				utime.sleep_ms(500)
				utime.sleep_ms(500)
				utime.sleep_ms(500)

		except OSError:
			if 'poller' in locals():
				poller.unregister(s)
		finally:
			sock.close()
			# Now set the status of the device to Not Connected
			iot_device_connected = False
`

The above is run inside an ESP8266. The latest 1.9.3 micropython is used. The above code connects to a server, and gets command from it. The main problem is that when the connection breaks (server process gets terminated), still I never receive POLLHUP or POLLERR.

Is there any mistake in my above code?

dpgeorge, 2018-05-01:
Sockets on esp8266 don't return POLLHUP or POLLERR at all.  If a connection is broken then the socket will become readable and a read on it will return `b''`.  So check for this return value and use that to detect a closed connection.

dpgeorge, 2018-05-05:
> The fd however when printed prints "socket state=3, ..."

state=3 is STATE_PEER_CLOSED, and such a socket is both readable and writable, that's how it works.

In commit 318f874cda22567a55bb004434ed0ec891fd7d61 logic was added to handle some cases when the socket is in an error state (state<0) and return POLLHUP/POLLERR, but this won't help with the code listed in the original post, because, as mentioned, that has a socket in STATE_PEER_CLOSED.

closed 2018-05-11 by dpgeorge