micropython/micropython #3749: Any hints on SPIRAM supports?

eminarcissus, 2018-05-01:
Hi,

I'm trying to migrating to support the 4M version with SPIRAM supports from labo's repo, from his change I didn't noticed much change in init process, just added SPIRAM supports with correct heap setting, but when it starts to booting up, micropython does crashes in mp_init function, with a error log like this
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:4336
ho 0 tail 12 room 4
load:0x40078000,len:0
load:0x40078000,len:10992
entry 0x4007a6c4
I (352) spiram: SPI RAM mode: flash 40m sram 40m
I (352) spiram: PSRAM initialized, cache is in normal (1-core) mode.
I (353) cpu_start: Pro cpu up.
I (357) cpu_start: Single core mode
I (1211) spiram: SPI SRAM memory test OK
I (1211) heap_init: Initializing. RAM available for dynamic allocation:
I (1212) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (1218) heap_init: At 3FFC5DA8 len 0001A258 (104 KiB): DRAM
I (1224) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (1230) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1237) heap_init: At 40090AF0 len 0000F510 (61 KiB): IRAM
I (1243) cpu_start: Pro cpu start user code
I (1248) spiram: Adding pool of 4096K of external SPI memory to heap allocator
I (1256) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (159) cpu_start: Starting scheduler on PRO CPU.
USE MALLOC with size 2097152
MALLOC with result 0x3f940020
W (188) MP: start task
W (188) WP: start thread
W (188) WP: start uart
W (198) WP: soft reset
W (198) WP: set stack
W (198) WP: set stack limit
W (208) WP: start gc
start mp nowNLR jump failed, val=0x3ffc3870
Guru meditation errors........

This is my setting parts on main function

`void app_main(void) {
    nvs_flash_init();
    // ==== Allocate heap memory ====
    #if CONFIG_SPIRAM_SUPPORT
    // ## USING SPI RAM FOR HEAP ##
    #if CONFIG_SPIRAM_USE_CAPS_ALLOC
    printf("USE CAPS ALLOC.\n");
    mp_task_heap = heap_caps_malloc(MP_TASK_HEAP_SIZE, MALLOC_CAP_SPIRAM);
    #elif CONFIG_SPIRAM_USE_MEMMAP
    printf("USE MEMMAP.\n");
    mp_task_heap = (uint8_t *)0x3f800000;
    #else
    printf("USE MALLOC with size %d\n",MP_TASK_HEAP_SIZE);
    mp_task_heap = malloc(MP_TASK_HEAP_SIZE);
    printf("MALLOC with result %p\n",mp_task_heap);
    #endif
    #else
    // ## USING DRAM FOR HEAP ##
    mp_task_heap = malloc(MP_TASK_HEAP_SIZE);
    #endif

    if (mp_task_heap == NULL) {
        printf("Error allocating heap, Halted.\n");
        return;
    }
    periph_module_disable(PERIPH_I2C0_MODULE);
    periph_module_enable(PERIPH_I2C0_MODULE);

    xTaskCreateStaticPinnedToCore(mp_task, "mp_task", MP_TASK_STACK_LEN, NULL, MP_TASK_PRIORITY,
                                  &mp_task_stack[0], &mp_task_tcb, 0);
}`

This is my mp_task part,
`
    ESP_LOGW("MP","start task");
    volatile uint32_t sp = (uint32_t)get_sp();
    #if MICROPY_PY_THREAD
    ESP_LOGW("WP","start thread");
    mp_thread_init(&mp_task_stack[0], MP_TASK_STACK_LEN);
    #endif
    ESP_LOGW("WP","start uart");
    uart_init();

soft_reset:
    // initialise the stack pointer for the main thread
    ESP_LOGW("WP","soft reset");
    mp_stack_set_top((void *)sp);
    ESP_LOGW("WP","set stack");
    mp_stack_set_limit(MP_TASK_STACK_SIZE - 1024);
    ESP_LOGW("WP","set stack limit");
    gc_init(mp_task_heap, mp_task_heap + sizeof(mp_task_heap));
    ESP_LOGW("WP","start gc");
    mp_init();
    ESP_LOGW("WP","start mp init");
    mp_obj_list_init(mp_sys_path, 0);
    ESP_LOGW("WP","set sys path");
    mp_obj_list_append(mp_sys_path, MP_OBJ_NEW_QSTR(MP_QSTR_));
    mp_obj_list_append(mp_sys_path, MP_OBJ_NEW_QSTR(MP_QSTR__slash_lib));
    mp_obj_list_init(mp_sys_argv, 0);
    ESP_LOGW("WP","start read line");
    readline_init0();
    ESP_LOGW("WP","init pins");
`

I trying to debug from mp_init function,
but since it builts with mpy_cross, I didn't link it with esp-idf, so it doesn't have a correct print function, and I compared gccollect and all other settings might caused the problem, all failed here.

Any hints on the root of the problem here?(or even how to debug on this bug here...)

dpgeorge, 2018-05-02:
I did get SPIRAM working myself, it just needed changes to the Makefile and sdkconfig.h, no changes to any C code.  The diff is:
```diff
diff --git a/ports/esp32/Makefile b/ports/esp32/Makefile
index eb0a3be24..8038a1837 100644
--- a/ports/esp32/Makefile
+++ b/ports/esp32/Makefile
@@ -255,6 +255,8 @@ ESPIDF_ESP32_O = $(addprefix $(ESPCOMP)/esp32/,\
        wifi_init.o \
        wifi_internal.o \
        sleep_modes.o \
+       spiram.o \
+       spiram_psram.o \
        )
 
 ESPIDF_HEAP_O = $(addprefix $(ESPCOMP)/heap/,\
diff --git a/ports/esp32/sdkconfig.h b/ports/esp32/sdkconfig.h
index 113c0395f..03d597815 100644
--- a/ports/esp32/sdkconfig.h
+++ b/ports/esp32/sdkconfig.h
@@ -46,6 +46,18 @@
 #define CONFIG_ESP32_PTHREAD_TASK_PRIO_DEFAULT 5
 #define CONFIG_ESP32_PTHREAD_TASK_STACK_SIZE_DEFAULT 2048
 
+/* SPIRAM support, will not work if the target doesn't have SPIRAM */
+#define CONFIG_SPIRAM_SUPPORT 1
+#define CONFIG_SPIRAM_TYPE_ESPPSRAM32 1
+#define CONFIG_SPIRAM_SIZE 4194304
+#define CONFIG_SPIRAM_SPEED_40M 1
+#define CONFIG_SPIRAM_CACHE_WORKAROUND 1
+#define CONFIG_SPIRAM_BOOT_INIT 1
+#define CONFIG_SPIRAM_MEMTEST 1
+#define CONFIG_SPIRAM_USE_MALLOC 1
+#define CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL 16384
+#define CONFIG_SPIRAM_MALLOC_RESERVE_INTERNAL 32768
+
 #define CONFIG_FOUR_MAC_ADDRESS_FROM_EFUSE 1
 #define CONFIG_DMA_RX_BUF_NUM 10
 #define CONFIG_DMA_TX_BUF_NUM 10
```
The reason I didn't push this change yet is because it will break boards that don't have SPIRAM.  So we need a way to select and configure individual boards, like the stm32 port does.

eminarcissus, 2018-05-02:
Hmm, kind of weird, I have that changed in my sdkconfig as well, I thought
that only enables SPIRAM in os level, I still need to change the initialize
stage. Will test it out, thanks very much for the info :)
-- 
Narcissus Emi

:

eminarcissus, 2018-05-02:
I build it with the latest micropython branch, but it still failed at boot stage.

I (360) spiram: SPI RAM mode: flash 40m sram 40m
I (360) spiram: PSRAM initialized, cache is in normal (1-core) mode.
I (361) cpu_start: Pro cpu up.
I (365) cpu_start: Single core mode
I (1218) spiram: SPI SRAM memory test OK
I (1219) heap_init: Initializing. RAM available for dynamic allocation:
I (1219) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (1225) heap_init: At 3FFC50A8 len 0001AF58 (107 KiB): DRAM
I (1232) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (1238) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1245) heap_init: At 4008A47C len 00015B84 (86 KiB): IRAM
I (1251) cpu_start: Pro cpu start user code
I (1256) spiram: Adding pool of 4096K of external SPI memory to heap allocator
I (1264) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
ru Meditation Error: Core  0 panic'ed (IllegalInstruction)
. Exception was unhandled.
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction)
. Exception was unhandled.

dpgeorge, 2018-05-02:
Do you have the latest xtensa-esp32 toolchain installed?  Check the version.

eminarcissus, 2018-05-02:
xtensa-esp32-elf-linux64-1.22.0-80-g6c4433a-5.2.0.tar.gz
Is this the latest one should be used here? This is the toolchain I have on my building machine.

dpgeorge, 2018-05-02:
Yes that's the latest, but double check by executing `xtensa-esp32-elf-gcc --version`.  Apart from that, I'd reclone both the ESP IDF repo and this uPy repo and try again.

eminarcissus, 2018-05-02:
xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Yes, I think so, really appreciated :)

eminarcissus, 2018-05-02:
Looks like I've got a faulty hardware.....
I have it flashed in other chip and it works like a charm... don't know where's the problem is.
will close it now, thx very much for the help

closed 2018-05-02 by eminarcissus