micropython/micropython #3751: del - add option to call gc_free()?

adritium, 2018-05-01:
```
a = create_some_obj()
del a
```

`del` doesn't free the memory in the heap; it merely removes `a` from the local dictionary and marks the pointer bound to `a` in the local dictionary as NULL . . . so the `gc` cleans it up.

In RAM-constrained environments, it's helpful to be able to clean up your heap sooner than later especially to reduce fragmentation.
 
And in performance-constrained environments, it's helpful to not need to call the `gc` to do the heap cleaning if it can be avoided due to the runtime overhead.

**REQUEST**

Add optional argument to `del` that to call `m_free` on the pointer held by the first argument of `del`

dhylands, 2018-05-01:
As mentioned on the forum, in general you del can't call free. For example:
```
a = create_some_obj()
b = [a]
del a
```
In this scenario if del called free then b would be invalid.

dpgeorge, 2018-05-02:
@dhylands is right, you can't do this in general, it will badly break Python semantics.

> And in performance-constrained environments, it's helpful to not need to call the gc to do the heap cleaning if it can be avoided due to the runtime overhead.

In cases that you really need to do this then at least `gc.collect()` is there, and the overhead of calling it is minimal.

> Add optional argument to del that to call m_free on the pointer held by the first argument of del

`del` is a statement with special syntax and there's no way we can modify the grammar to make it accept an "optional argument".  The syntax `del a, b` already exists and means to delete both a and b.

adritium, 2018-05-02:
@dpgeorge reference counting would solve this. Any plans to switch the `gc`?

dpgeorge, 2018-05-03:
> reference counting would solve this. Any plans to switch the gc?

No plans.  Reference counting requires a huge change to the code base, to sprinkle incref/decref in many locations, including in all extmod code, port code, and user-defined C extensions.

adritium, 2018-05-07:
@dpgeorge Why not add a "use at your own peril" function `gc.free_ref(a)` that calls `del` and then calls `free`? 

@dhylands 

I know that it may not be standard Py but with microcontrollers, it's expected you'll have memory issues so having a tool to deal with it makes sense to me.

stinos, 2018-05-07:
See comments above: del+free is principally wrong so adding it as a core function is a bad idea. If anthing if should be del+gc_collect, which you can easily write yourself. Or restructure the code so there aren't memory usage problems.  Also I'm not sre this really is that common of a problem, I don't think anyone else ever truly needed it, or at least I don't think it ever has been asked for here or on the forum.

adritium, 2018-05-07:
@stinos 

Show me how to

>  restructure the code so there aren't memory usage problems.

```
stored = []
while Condition():
    m1 = q1.get()
    stored.append(m1)    
    m2 = q2.get()
```

> I don't think anyone else ever truly needed it

You can say that about any feature and any invention.

stinos, 2018-05-07:
> Show me how to

This doesn't seem a proper code sample as m2 is unused. And I don't know what type q2 is. Hard to tell anything sensible about it.

> Really? "well in my work I haven't needed it so therefore no one in any situation needs it"
> You want to close off uPy to an entire industry because it doesn't align with your work?

Not sure how you got that idea, I never even hinted at that. My point simply is that in all the years I've been subscribed to uPy on Github and the forum, I do not recall anyone ever asking of this feature explicitly. Note that does not mean a) no one ever asked for it, just that *as far as I remember* no-one did, for all I know I'm wrong and missed it b) no one ever wanted it. But it does hint there's not a huge market for it since there's no trace of it.

> but there are exceptions when they are useful.

Sure it could be useful for your particular case. But again, there's so many things which are useful for a tiny percentage or even just one usecase it's not worth spending time on implementing them, writing tests, either increase code size or have yet another configuration option, etc., especially when they are also extremely tricky. However you could of course just keep it in your own fork like many others do.

adritium, 2018-05-07:
@stinos It's perfectly proper: an element is being removed from a queue.
It wouldn't make my point further if I added `doSomeThing(m2)`

You can't figure out what `q1` refers to? Say it aloud.

Neon22, 2018-05-07:
Ahh would it help to use a preallocated buffer for variable stored and then you'd be working in a stable memory config.
such as bytearray() (obviously for bytes but you get the idea - just don't use append().)
E.g. from https://forum.micropython.org/viewtopic.php?t=1259
```
>>> buf = bytearray('0123456789')
>>> buf[3:6] = b'abc'
>>> buf
bytearray(b'012abc6789')
>>> mv = memoryview(buf)
>>> bytes(mv)
b'012abc6789'
>>> mv[3:6] = b'abc'
>>> buf
bytearray(b'012abc6789')
```
Also if the object you're reading from (making a lot of assumptions here) is a stream then
objects which support stream interface (e.g. file or UART) provide read() method which allocates new buffer for read data, but also a readinto() method to read data into an existing buffer.
From: - http://docs.micropython.org/en/v1.9.3/wipy/reference/speed_python.html

implementing a circular buffer (no idea if you need one) would not be much additional code.

peterhinch, 2018-05-08:
As I understand it in assigning to a variable doesn't cause an allocation - after all you can do it in an ISR. I think a pointer is put on the stack.

stinos, 2018-05-08:
> @stinos It's perfectly proper: an element is being removed from a queue. What you do with it isn't relevant to the point I'm making regarding memory.

Is is relevant with respect to you asking how to restructure code. Based on the lack of context, just for your cdoe sample, I can come up with a myriad of solutions none of which you'd be able to use in the actual real life code. For example just based on that code, a perfect solution one could propose is to simple get rid of q2 all together, after all you're seemingly populating a queue just to remove items from it again without using them. *Seems* like a no-op and wasting memory for nothing, in reality it likely isn't..

> You can't figure out what q1 refers to? Say it aloud.

Let's remain polite, shall we. Also, 'a queue' doesn't tell which type, i.e. which implementation; uPy's? Some external library? Your own. Etc. This matters for proposing solutions like Neon's.

peterhinch, 2018-05-08:
My point above is that the line
```python
m2 = q2.get()
```
removes an object from a queue and creates a pointer to it. The pointer (as I understand it) lives on the stack. So there is no relevance in deallocating the pointer. If the aim is to deallocate the object to which it points, that seems a recipe for fragile code as there may be other references to that object. Debugging the resultant intermittent crashes would be a nightmare.

If such a facility existed I wouldn't consider using it. It really shouldn't be possible to crash the VM using Python code.

Safer to issue a `gc.collect()` every N iterations of the loop, which is guaranteed to deallocate the objects **if** there are no references to them which are in scope.

adritium, 2018-05-08:
@stinos @peterhinch my mistake. I am using the queue abstraction for my purposes but it's not really a Python queue. I'll update later.

adritium, 2018-05-09:
@peterhinch 

> It really shouldn't be possible to crash the VM using Python code.

That's already possible: use `machine.mem32` to write to the wrong address and your micro will reset.

But I guess that's not a fair example since we're comparing a core uPy language feature to the implementation of a library.

Implementing this **free** feature isn't hard so I'll probably just add it to my firmware.
I wanted to provoke discussion but I do see that I'm trying to turn Python into something it's not meant to be.

dpgeorge, 2018-05-10:
@adritium can you please refrain from heavily editing your comments after you post them?  Doing so makes it hard to follow the discussion.

> The author of this code knows that m2 will not be used in the next iteration loop.
> As the loop runs, the objects temporarily stored by m2 will become orphaned and eventually, the gc will activate.
> The problem is that the gc takes a huge amount of time, relative to rest of code, to execute and if this loop is timing critical-ish, it will introduce jitter.
> 
> The point I'm making is that since the user knows m2 will not be needed when the loop is done, why shouldn't the user be allowed to free m2 at their own risk?

It really depends on what kind of object m2 is as to whether the approach of explicitly freeing it would help.  If m2 is a composite object (ie uses more than one GC block in its storage) then it's not going to be freed simply by calling m_del() on it (which only frees the block that the pointer refers to).  It really needs a full GC to properly free it because some parts of it may be shared with other objects.

adritium, 2018-05-10:
> can you please refrain from heavily editing your comments after you post them? Doing so makes it hard to follow the discussion.

Yes, sorry.

> then it's not going to be freed simply by calling m_del() on it (which only frees the block that the pointer refers to).

Yes I see the issue.

It looks like the solution you've proposed in #3774 in [this comment](https://github.com/micropython/micropython/issues/3774#issuecomment-387963973) both satisfies my requirements and does so in a non-"you could crash the VM" manner.

I'll close this issue in a few days if no other contributions are made.

adritium, 2018-08-02:
I agree with commenters that adding this as a core function is a bad idea. I'll add it to my port.

closed 2022-06-23 by dpgeorge