micropython/micropython #3753: pypi.org - now requires SSL for package download - making upip.py fail

aenertia, 2018-05-01:
I encountered this issue on my esp8266 then attempted to replicate on a unix build of micropython; I understand that the cacert store is needed, but even on a fresh build of the unix port it will fail unless ca certs are copied into the build root.

The debug message should probably include some reference to the above, and since pypi archives now force ssl - it makes upip somewhat useless on small embedded platforms where ssl may or may not be available due to resource constraints.

dpgeorge, 2018-05-02:
upip does support and use SSL, it just doesn't validate the certificate.  And upip works OK for me using the latest unix version without ca certs.

Can you provide reproducible steps that shows your issue?

aenertia, 2018-05-02:
Compiled from git on Ubuntu18.04:
commit 60c6b880fa4077a05d1273606d41fdc33f06bc2b (HEAD -> master, origin/master, origin/HEAD)

myupip is upip with debug set to True
--
aenertia@kiorewha:~/build/micropython$ cd ports/unix/
aenertia@kiorewha:~/build/micropython/ports/unix$ ./micropython -m myupip install micropython-uasyncio
Installing to: /home/aenertia/.micropython/lib/
Queue: ['micropython-uasyncio']
https://pypi.python.org/pypi/micropython-uasyncio/json
Warning: pypi.python.org SSL certificate is not validated
Error installing 'micropython-uasyncio': Package not found, packages may be partially installed
--

I get similar result issues on the esp8266 (witty cloud dev board) build I am running which was built in Janurary.
--

>>> upip.install('uasyncio')                                                                                                              
Installing to: /lib/                                                                                                                      
Error installing 'uasyncio': Package not found, packages may be partially installed                                                       
>>> upip.install('micropython-uasyncio')                                                                                                  
Installing to: /lib/                                                                                                                      
Error installing 'micropython-uasyncio': Package not found, packages may be partially installed

aenertia, 2018-05-02:
Fetching the package URL from a browser is fine... fetching via wget with http results in a pypi.org message saying SSL Must be used.

```
aenertia@kiorewha:~/build/micropython/ports/unix$ wget http://pypi.python.org/pypi/micropython-uasyncio/json
--2018-05-02 13:15:11--  http://pypi.python.org/pypi/micropython-uasyncio/json
Resolving pypi.python.org (pypi.python.org)... 2a04:4e42:27::223, 151.101.164.223
Connecting to pypi.python.org (pypi.python.org)|2a04:4e42:27::223|:80... connected.
HTTP request sent, awaiting response... 403 SSL is required
2018-05-02 13:15:12 ERROR 403: SSL is required.
```

I wonder if this is something to do with ipv6 being present on the transit network perhaps?

aenertia, 2018-05-02:
Which branch should I be using? I built the Unix version from Git this morning from 'master' and it exhibits the issue above.

dpgeorge, 2018-05-02:
Using the master branch of this repository, with commit 6b4b6d388be338d36de01b01e25f6e324321750a I get:
```
$ ./micropython -m upip install micropython-uasyncio
Installing to: .micropython/lib/
Warning: pypi.org SSL certificate is not validated
Installing micropython-uasyncio 2.0 from https://files.pythonhosted.org/packages/9d/a4/afaf98be714d3701d47c47b3b6aa3d5e52bf277fe8ceb9a2e3fdb05eaf86/micropython-uasyncio-2.0.tar.gz
Installing micropython-uasyncio.core 2.0 from https://files.pythonhosted.org/packages/66/45/372e657680a613ecb6d4d124157fa46f99d2072b4443476a20f39a82f961/micropython-uasyncio.core-2.0.tar.gz
```

aenertia, 2018-05-02:
Confirmed working - I had issues with the local git repo; was pulling from
master and showing as up-to-date but clearly wasnt.
'''
aenertia@kiorewha:~/build/micropython/ports/unix$ ./micropython -m upip
install micropython-uasyncio
Installing to: /home/aenertia/.micropython/lib/
Warning: pypi.org SSL certificate is not validated
Installing micropython-uasyncio 2.0 from
https://files.pythonhosted.org/packages/9d/a4/afaf98be714d3701d47c47b3b6aa3d5e52bf277fe8ceb9a2e3fdb05eaf86/micropython-uasyncio-2.0.tar.gz
Installing micropython-uasyncio.core 2.0 from
https://files.pythonhosted.org/packages/66/45/372e657680a613ecb6d4d124157fa46f99d2072b4443476a20f39a82f961/micropython-uasyncio.core-2.0.tar.gz
'''

closed 2018-05-02 by dpgeorge

ponyatov, 2019-07-10:
MicroPython 09a492b-dirty on 2019-07-10; linux version

dpon@dpon:~/KBMS$ bin/micropython -m upip install ply
Installing to: /home/dpon/.micropython/lib/
Warning: micropython.org SSL certificate is not validated
Error installing 'ply': , packages may be partially installed
dpon@dpon:~/KBMS$ 

dpon@dpon:~/KBMS$ find ~/.micropython
/home/dpon/.micropython
/home/dpon/.micropython/lib
/home/dpon/.micropython/lib/network.py
/home/dpon/.micropython/lib/socket.py