micropython/micropython #3756: Failed to boot with error message - Image contains multiple IROM segments(On ESP32 chip)

eminarcissus, 2018-05-02:
Hmm.
Another weird problem here,

This is the boot log.
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:4436
load:0x40078000,len:0
load:0x40078000,len:11816
entry 0x4007a9fc
E (514) boot: Image contains multiple IROM segments. Only the last one will be mapped.

This is the build log for firmware
LINK build/application.elf
   text	   data	    bss	    dec	    hex	filename
 957334	 295205	  99152	1351691	 14a00b	build/application.elf
Create build/application.bin
esptool.py v2.2.1
Create build/firmware.bin
bootloader     16352
partitions      3072
application  1363664
total        1429200

dpgeorge, 2018-05-03:
Sorry, can't help here.  This is not really a bug/issue with MicroPython, you'll need to understand how the ESP IDF works to fix it.

eminarcissus, 2018-05-03:
Got it, will get to ESP IDF to see the reason here, I've checked IDF's document for quite a lot of times but didn't get any insights about this problem. Thx very much for the help

closed 2018-05-03 by eminarcissus