micropython/micropython #3759: ESP8266 SPI slave mode

Fef0, 2018-05-04:
Hi, I'm new to the micropython world and I'm actually working with an ESP8266 which has to act as a SPI bridge between two SPI devices, is there any way to use the Slave mode on it? I tried with the library which is included with micropython but I can't see any slave mode.
Some ideas?

peterhinch, 2018-05-05:
Slave mode is currently unsupported. An implementation on the ESP8266 capable of any reasonable speed would probably be difficult because of latency issues.

Fef0, 2018-05-05:
My system should communicate at 9600 Thanks for lbaud, so not a big deal.. What about the ESP32? Is it supported?

peterhinch, 2018-05-06:
The same observation applies: ESP32 and ESP8266 both have an underlying RTOS. Slave mode with reasonable performance is probably impracticable except on bare-metal ports such as the Pyboard. The problem is that the interface assumes a fast realtime response from the slave.

Fef0, 2018-05-06:
Okay thanks for the clarification Peter!

Fef0, 2018-05-07:
@dpgeorge Thanks for the clarification, I was thinking to use two ESP8266: one as slave (to receive data from the original master) and one as master (to control the original slave with the edited data), both connected with serial or I2C to exchange data. I could use an Arduino, but I need a fast computation on the bridge node, so an ESP8266 at 160MHz would be perfect. I read that the ESP8266 has a hardware SPI port, but I couldn't understand why micropython exposes just the master mode. I also have a ESP32 which should have 2 SPI port, one called HSPI and another called VSPI, so the slave-master nodes would be condensed in a single node

mattytrentini, 2024-09-17:
Issue appears resolved, at least until an SPI Slave implementation is more widely available.

closed 2024-09-17 by mattytrentini