micropython/micropython #3760: uclinux (uclibc 0.9) endianness detection fails on big endian arch (microblaze)

dxxb, 2018-05-05:
On this arch checking for the existence of `__LITTLE_ENDIAN` and `__BIG_ENDIAN` breaks endianness detection.

`__LITTLE_ENDIAN` and `__BIG_ENDIAN` (two underscore prefix) are both defined but `__LITTLE_ENDIAN` is checked first here https://github.com/micropython/micropython/blame/master/py/mpconfig.h#L1266. The current mechanism was introduced by https://github.com/micropython/micropython/commit/a9bcd51dc7f05be0961f699f39d1856e78eaaaa6 and https://github.com/micropython/micropython/commit/f32498fe04c959e9696d5409adc33f1f518ec76e

Defined endianness related macros:
```
#define __FLOAT_WORD_ORDER __BYTE_ORDER
#define BYTE_ORDER __BYTE_ORDER
#define __BYTE_ORDER __BIG_ENDIAN

#define __LITTLE_ENDIAN 1234
#define __BIG_ENDIAN 4321
#define __PDP_ENDIAN 3412

#define _ENDIAN_H 1
#define PDP_ENDIAN __PDP_ENDIAN
#define BIG_ENDIAN __BIG_ENDIAN
#define LITTLE_ENDIAN __LITTLE_ENDIAN

#define __ARCH_SUPPORTS_BIG_ENDIAN__ 1
#define __ARCH_BIG_ENDIAN__ 1
#define _BIG_ENDIAN 1
```

dpgeorge, 2018-05-05:
The autodetection tries to do its best but that's not always going to work for every system.  There's always the option to specify the endianness explicitly by defining `MP_ENDIANNESS_BIG`.  Which port are you compiling (unix or some custom port)?  Are you able to explicitly specify `MP_ENDIANNESS_BIG`?

Does the system define the macro `__ORDER_LITTLE_ENDIAN` or something similar?  Because that might be used to improve the autodetection.

dxxb, 2018-05-05:
> Does the system define the macro __ORDER_LITTLE_ENDIAN or something similar?

No, all endianness related macros are listed in my opening comment.

> Which port are you compiling (unix or some custom port)?

unix

> Are you able to explicitly specify MP_ENDIANNESS_BIG?

Yes, and it works, but it is inconvenient because it means I cannot compile the same source code for testing on a different arch.

> The autodetection tries to do its best but that's not always going to work for every system.

I understand it is tricky and that it may not be possible to fix easily. However I am wondering if the `defined(__BIG_ENDIAN)` and `defined(__LITTLE_ENDIAN)` belong in the #if statements because the mere existence of (say) `__BIG_ENDIAN` may not indicate the architecture is big endian.

dpgeorge, 2018-05-07:
Please see #3771 for a PR which aims to fix this issue.

dpgeorge, 2018-05-07:
> However I am wondering if the defined(__BIG_ENDIAN) and defined(__LITTLE_ENDIAN) belong in the #if statements because the mere existence of (say) __BIG_ENDIAN may not indicate the architecture is big endian.

Yes I agree this seems wrong.  That's what #3771 tries to address, hopefully it doesn't break existing systems...

dxxb, 2018-05-07:
Thank you @dpgeorge. I tried #3771 and it works.

dpgeorge, 2018-05-08:
Thanks for confirming.  I'll wait until after v1.9.4 is released before merging #3771, because it may have some as-yet-unreported problems.

dpgeorge, 2018-05-11:
Fixed by 9630376dbca0202321f2aae6ef88ef75cdb5374e

closed 2018-05-11 by dpgeorge