micropython/micropython #3762: Unable to access python console using picocom

lazarvgd, 2018-05-05:
Hi there,
I have tried to flash my esp8266(LoLin board) so I can use python on it. And I made it with 2 boards without a problems. But I have tried another 2 and got problem...

The problem:
I have execute these commands

`sudo esptool.py --port /dev/ttyUSB1 erase_flash`
and this one
`sudo esptool.py --port /dev/ttyUSB1 --baud 460800 write_flash --flash_size=detect 0 ~/Downloads/esp8266-20171101-v1.9.3.bin`

and finally tried to access with this command

`sudo picocom -b 115200 /dev/ttyUSB1`

and I am getting picocom console, and if I hit ENTER btn nothing happens, no python consol sign (">>>")
I have found this [bug](https://github.com/micropython/micropython/issues/1895), but I do not think this could be related to my problem, due to I have 2 boards with working python on it.
Also I have tried to upload code to these non-working esp's from arduino ide and I done it without problems.

Does anybody knows what I can do.

dpgeorge, 2018-05-07:
> And I made it with 2 boards without a problems. But I have tried another 2 and got problem...

Are these other 2 non-working boards exactly the same as the 2 working boards?  Even if they are the same brand/model they may have different flash memory, which can make a difference.

How much flash memory do the boards have?  Did you try changing the flash mode from qio to dio (`--flash_mode=dio`)?

lazarvgd, 2018-05-07:
To be honest, I don't know, they look the same, can you please let me know how can I check the memory? I have not tired changin anything, due to lack of knowladge (not very familiar with these things - just started).

dpgeorge, 2018-05-08:
> After adding the flag --flash_mode=dio I got my boards working!
> 
> Can you please explain me what was wrong with my approach?

flash-mode is how the SPI flash memory communicates with the esp8266 chip.  The default is qio which uses 4 (quad) lines and is the fastest.  dio uses 2 (dual) lines.  Different SPI flash chips have different commands for using qio mode, and some esp8266 boards may not even connect all 4 lines up to the SPI flash.  So that means different esp8266 boards may or may not work with the qio setting.  Whether a particular board works with qio is sometimes just up to trial and error, which you have to do yourself.

The only thing wrong with your approach was that you didn't try the dio mode.  It is mentioned in the tutorial: http://docs.micropython.org/en/latest/esp8266/esp8266/tutorial/intro.html#deploying-the-firmware

closed 2018-05-08 by dpgeorge