micropython/micropython #3763: REPL hangs after UART initializing

alessss, 2018-05-05:
Hello!
I'm trying to read mhz19 co2 sensor data using ESP12F with micropython.

My pins are connected as following: 
MHZ19    ESP
GND --> GND
Vin    --> Vin 
Tx     -->  Rx
Rx    -->  Tx

Console hangs directly after connecting (screen /dev/cu.wchusbserial14340 115200) giving following output:

`��'�ng�l��d�l��l$������d�g����cll�;x���cl�#{$s$s�'�����g����c��$�d���cl���<cd`��lg�p;l�l��|��{�#4 ets_task(40100130, 3, 3fff837c, 4)
OSError: [Errno 2] ENOENT

MicroPython v1.9.3-8-g63826ac5c on 2017-11-01; ESP module with ESP8266
Type "help()" for more information.
>>> `

Firmware without mhz19 sensor works well. What's the problem with the port?

robert-hh, 2018-05-06:
If you connect something else to RX and TX, you'll have a collision with the console, which is also connected to RX and TX. The collision is both for the electrical signals and the protocol, In this case, the mh-z19 sensor requires a setting of 9600 baud, whereas the console is using 115200 baud.
I faintly remember some discussion of setting the console to webrepl only, but could not find that any more.

dpgeorge, 2018-05-07:
The UART on esp8266 is currently only available for the REPL and can't be used for other things.

alessss, 2018-05-07:
http://micropython.org/webrepl/ - can webrepl help me in my case?

dpgeorge, 2018-05-08:
> - can webrepl help me in my case?

Not until it's possible to disable the REPL on the UART; see #3277 for some code which does this.

dpgeorge, 2018-05-21:
It's now possible to disable the REPL on UART(0) on the esp8266, see afd0701bf7a9dcb50c5ab46b0ae88b303fec6ed3.

closed 2018-05-21 by dpgeorge

johnwebsiteprojects, 2025-01-19:
i just hoped i saw this solution before frying my chip 

> http://micropython.org/webrepl/ - can webrepl help me in my case?