micropython/micropython #3764: ESP32 - External 32.768 Xtal

diginfo, 2018-05-06:
I understand that the IDF / ESP supports an external crystal for the built in RTC on pins GPIO32 & GPIO33.

Can this be enabled on this ESP32 build ?

Thanks

dpgeorge, 2018-05-07:
It should be as simple as adding this line to esp32/sdkconfig.h:
```
#define CONFIG_ESP32_RTC_CLOCK_SOURCE_EXTERNAL_CRYSTAL 1
```

dmartauz, 2019-02-28:
@diginfo Could you please confirm that it is possible to make use of external RTC crystal by using this build setting?

dpgeorge, 2019-03-01:
Since the esp32 port changed to use the Kconfig sdkconfig file, the way to do this now is to add the following line to your custom board/sdkconfig file:
```
CONFIG_ESP32_RTC_CLOCK_SOURCE_EXTERNAL_CRYSTAL=1
```

dmartauz, 2019-03-01:
Small correction, as bool parameter it must be:
`CONFIG_ESP32_RTC_CLOCK_SOURCE_EXTERNAL_CRYSTAL=y`

dpgeorge, 2019-06-24:
As there's a solution for this (at the user's end) I'll close the issue.

closed 2019-06-24 by dpgeorge