micropython/micropython #3765: ESP32 - Uart Buffer Size

diginfo, 2018-05-06:
The buffer size seems really small on the ESP32, is it possible to increase this or preferably make it configurable ?

dpgeorge, 2018-05-07:
It's possible to increase it in your own custom build by changing the values on this line: https://github.com/micropython/micropython/blob/master/ports/esp32/machine_uart.c#L243

Eventually it would make sense to make it dynamically configurable with keyword arguments to the UART's constructor, like `rx_buf_len` and `tx_buf_len`.

diginfo, 2018-05-10:
The loboris port uart buffer receive size is configurable and defaults to 1024 bytes, however I am unable to use it reliably as there are frequent 'guru mediation' errors and 'cpu halted' and these are show-stoppers when you want a device to be dependable & standalone.

dpgeorge, 2018-12-05:
Ability to configure the esp32's UART rx/tx buffer sizes was implemented in 9ddc182ec7195722de426d5461ed1d70d9aedf7f

closed 2018-12-05 by dpgeorge