micropython/micropython #3769: accessing c structs of parent classes (e.g. subclassing Pin)

distillcode, 2018-05-06:
I'm creating classes implemented in C representing various hardware elements. In the constructor (in C) I want to access elements from the c structure of an object passed in. This is fine if I pass in an instance of an object whose class I defined in C. However, if I pass an object that is an instance of a class (defined in Python) that inherits from the C defined class, can I access the c structure of the parent somehow?

As a more concrete example, I think same problem would exist related to the Pin object  e.g.

```
class OnlyInputPin(Pin):
    pass

derived_pin = OnlyInputPin(3)
normal_pin = Pin(2)

ADC(derived_pin)
```

and more specifically, looking at the ESP32 implementation of an adc and how it interacts with Pin.

The Pin class definition in C contains:
 
```
gpio_num_t machine_pin_get_id(mp_obj_t pin_in) {
    if (mp_obj_get_type(pin_in) != &machine_pin_type) {
        mp_raise_ValueError("expecting a pin");
    }
    machine_pin_obj_t *self = pin_in;
    return self->id;
}
```

So when a adc is instantiated, in the following c:

```
STATIC mp_obj_t madc_make_new(const mp_obj_type_t *type, size_t n_args, size_t n_kw,
        const mp_obj_t *args) {
...
    mp_arg_check_num(n_args, n_kw, 1, 1, true);
    gpio_num_t pin_id = machine_pin_get_id(args[0]);
```

It appears to need to have an object machine_pin_type passed into it, and this object can't be of a type that 'inherits' from machine_pin_type. 

In summary, is there an easy way to get the c structure of an object that is a subclass. i.e. in the Pin example, can I get a pointer to the machine_pin_obj_t from a reference to an object which is an instance of the OnlyInputPin class created in python? 

Given I can't find an example of anywhere this is done, Is this related to #2919 in that it's not an easy thing to do in micropython?

dpgeorge, 2018-05-07:
> In summary, is there an easy way to get the c structure of an object that is a subclass. i.e. in the Pin example, can I get a pointer to the machine_pin_obj_t from a reference to an object which is an instance of the OnlyInputPin class created in python?

Yes, actually, you can use the helper function `mp_instance_cast_to_native_base()`.  Eg:
```c
mp_obj_t pin_o = mp_instance_cast_to_native_base(pin_in, MP_OBJ_FROM_PTR(&machine_pin_type));
```
If it returns `MP_OBJ_NULL` then it wasn't of the correct type.

distillcode, 2018-05-18:
Thanks, I was having trouble with this, but it turned out to be a problem elsewhere. Many thanks for your help.

closed 2018-05-18 by distillcode