micropython/micropython #3770: Enhancement - add sub() to re module

diginfo, 2018-05-06:
Hi;

This is a very useful function for parsing strings, any chance it can be aded to the re module ?

dpgeorge, 2018-05-07:
Yes this would be useful to have, but how to implement it in a "micro" way is what needs to be understood.

For anyone who would find this to be a good addition, please vote on this issue.

diginfo, 2018-05-23:
Hi Damien;

Is the code @dmazzella provided good / small enough to include these functions ?

dpgeorge, 2018-05-24:
I cleaned up the patch provided by @dmazzella and made it a PR in #3815.

diginfo, 2018-05-24:
That's really great, thanks so much.

dmazzella, 2018-05-24:
I have commented your PR #3815

diginfo, 2018-06-12:
Sorrie to pester but...

is it possible to get s 'pre-release' of this as I find myself in need of it again to compress (remove whitespaces) from a html page ?

It looks like it failed some tests here: https://github.com/micropython/micropython/pull/3815

stinos, 2018-06-12:
You can fetch the code from the PR like this: ```git fetch upstream pull/3815/head:pr/3815``` then ```git checkout pr/3815```
For reuse make this a git alias like ```!sh -c 'git fetch upstream pull/'$1'/head:pr/'$1'```

dpgeorge, 2018-06-13:
> It looks like it failed some tests here:

I've fixed the PR so all tests pass, all new lines of code are covered by new tests.

dpgeorge, 2018-07-02:
ure.sub() was added in e30a5fc7bcd27900e0657db97ed54fc056d8f852, and enabled on stm32, esp8266, esp32 in 8f86fbfd6c34a4d03f2bd62e9dc1ff59c236b037

closed 2018-07-02 by dpgeorge