micropython/micropython #3774: add syntax to hint to gc lifetime of variable

adritium, 2018-05-07:
On PC, memory is almost a non-issue but with uPy running on microcontrollers, issues with heap - fragmentation, garbage collection runtime overhead - must sometimes (?often) be taken into consideration by the `.py` author. 

[#2057: Failure to allocate when free memory is very large - severe fragmentation?](#2057)

There are efforts to ameliorate some heap-issues: issue #3586 introduces concept of long- vs short-lived heap to reduce fragmentation.

However, these efforts are leaving out a critical piece of info: the context of the code in `.py`

My personal example:
```
long_term_msgs = []
while True:
    # m1 - stored "long-term"
    m1 = q1.get()
    # m2 - only exists for duration of while loop    
    m2 = q2.get()     
    long_term_msgs .append(m1)
```

The author knows that `m2` is a "temp" so it would be nice to be able to indicate this to the memory management system to make use of this info.

Yes, `.get()` could be modified with an arg to indicate the temporary nature but that seems messy: every function that allocates memory in C - even ones that are not obvious to a uPy-only author - would have to have this "isTemp" flag added.

A better way is to introduce syntax to indicate this.

```
long_term_msgs = []
while True:
    # m1 - stored "long-term"
    m1 = q1.get()
    # something like this - this is the "signal" that "m2" is meant to be short-term
    @micropython.temp:
    # m2 - only exists for duration of while loop    
        m2 = q2.get() 
    long_term_msgs.append(m1)
```

And how would it be implemented?
I'm thinking: `@micropython.temp` serves as a context which results in a C-call to switch a _malloc_ function pointer between 
`+#define m_new(type, num) ((type*)(m_malloc(sizeof(type) * (num), false)))`
and
`+#define m_new_ll(type, num) ((type*)(m_malloc(sizeof(type) * (num), true)))`

See [commit link](https://github.com/adafruit/circuitpython/commit/416abe33ed44e44d8f75342731080217e123278b#diff-4020fb3b71953b1afd39bd72ec54e68d) in #3586

peterhinch, 2018-05-08:
That syntax would break CPython compatibility. Is it possible for the compiler to deduce which variables have a short lifetime and use the short-lived heap?

adritium, 2018-05-08:
@peterhinch 

> That syntax would break CPython compatibility. 

I'm not an expert on syntax; I'm not saying it _must_ be what i wrote that, just **something** to indicate to the compiler/`gc` that the ?upcoming? allocations are of the "temporary" variety.

peterhinch, 2018-05-09:
If https://github.com/micropython/micropython/issues/3586 is implemented it's a good idea - assuming the compiler needs help in determining the lifetime of an object. The hint needs to occur when an object constructor is invoked, including the instantiation of built-in types. The semantics of a hint differs from a decorator: a decorator takes effect when a function is declared rather than when it is called.

The difficult part is devising a syntax for this which does not break CPython compatibility. An example is the **const** keyword where CPython compatibility can be retained by means of a simple lambda.

Incidentally editing posts after others have quoted from them makes threads hard to follow.

adritium, 2018-05-09:
> Incidentally editing posts after others have quoted from them makes threads hard to follow.

Noted. I'll just append instead of edit from now on.

> The hint needs to occur when an object constructor is invoked, including the instantiation of built-in types.

If you're saying apply the "hint" **only** when the `.make_new` function is invoked . . . I see your logic and I definitely see that limiting this hint to the constructor would make this task easier and I'd be happy if this was the first draft. 

However, I would really hope the hint could be applied anywhere because allocations can theoretically happen in any function.
From my example:

> m1 = busObj1.AllocateAndReturn()

`busObj1` represents a specific bus and channel i.e. CAN bus 1 or CAN bus 2. `AllocateAndReturn()` aka `get()` is a method of the class `busObj`. 
If this hint only applies to `.make_new`, the syntax would have to become weirder.

`m1 = busObj.message(busObj1)` 

where `busObj` is the type of `busObjX`, `message` is the constructor of the message format that is supported by `busObj` and we'd have to pass in the specific bus from which to pull the message.

With both syntaxes, you're calling a function so I don't see why it matters whether you use the first or second version (though I prefer the first). 
It'd be up to me, the creator of this API, to document whether a function performs an allocation (that can be influenced by a hint) and up to the `.py` author to determine whether his memory constraints warrant using this hint. 

As to how to implement this hint, the least intrusive way (I think) is for this hint to resolve to a C-call to flip some function pointer, before and after the marked function in the `.py` code, between these two

+#define m_new(type, num) ((type*)(m_malloc(sizeof(type) * (num), false)))
and
+#define m_new_ll(type, num) ((type*)(m_malloc(sizeof(type) * (num), true)))

In all C-functions of a class that perform allocations AND where you want to give `.py` author the option to control the allocation, `m_new_obj_with_finaliser`, etc would be replaced with macros that point to the flippable func pointer.

peterhinch, 2018-05-10:
Very nice. Limiting the returned variable to a single machine word (as in your example) seems quite a severe restriction. Is there any possibility of copying allocated objects onto the caller's heap on return? This would enable objects such as floats to be returned.

I guess the workround would be to use a method and bound variables or even globals.

dpgeorge, 2018-05-10:
> Is there any possibility of copying allocated objects onto the caller's heap on return?

That would be really good if possible.  I think it's possible, but might need to be restricted to a subset of basic datatypes (eg bignum, float, tuple, list, dict, string; and nested versions of these) because to copy an object you need to know its exact layout in memory, and then have special code to copy each kind of object.  It would act like a "compacting collector", sweeping the entire scoped-heap and copying out just the return value to the caller's heap (which could also be scoped).

> I guess the workround would be to use a method and bound variables or even globals.

No, even that won't work.  You won't be able to store any newly created objects from a scoped section into globals, because these objects won't exist once the scope is left.  One way around this would be to use a copying function like mentioned above, and exposing it as a function, eg: `some_global = micropython.promote_to_heap(some_object_allocated_on_scoped_heap)`.  Even better would be to do this automatically, for the runtime to detect when you write an object to the outer scope and promote the object, but that requires a lot of deep changes (eg write barriers for all stores of uPy objects).

Having a scoped heap is kind of like having the ability to do a contained soft-reset within your code.  Eg you'd wrap your main processing loop in a scoped heap and it would run the same, every time, without any memory fragmentation (of course with limitations, as discussed above).

adritium, 2018-05-10:
> 200 bytes are allocated on the C stack and used to create a temporary "heap"

This concerns me: why does `.py` author have to explicitly know how much heap to allocate? That seems like way too much info for a `.py` author to need to know. All they care about is _all allocations in this function will go away_ The `.py` author and firmware author could be separate people; I don't want my customer's scripts to break because a firmware change resulted in **200 bytes** not being enough.

I get that if it's on the stack, it's easier to implement but does it have to be on the stack?
Why can't the temp-heap be located in the already allocated heap? Couldn't you do some extra accounting to keep track of temp-heap objects so the function "destructor" knows which blocks to free? I'm guessing you'd have to modify this section (to start):
https://github.com/micropython/micropython/blob/3678a6bdc6a925f2bce6423c41f911229836f946/py/gc.c#L60-L64
When you allocate "temp-heap" objects, you'd mark it with a different `#define AT_HEAD/TAIL_TEMP` macro and then when the temp-heap context exits . . . you don't have to do a full GC but merely loop through the heap and mark any `#define AT_HEAD/TAIL_TEMP` as free. 
You could do this even faster if you kept an index of the temp-heap blocks so you avoid looping through entire heap (at the cost of some memory).
This way, we keep everything dynamic and no `.py` author is required to hardcode a heap amount.

And of course all this has to be reconciled with #3586: long-lived v short-lived heap to _prevent fragmentation_. 3586 doesn't have a mechanism to accept hints from the `.py` author as it relies on the judgement of the C-module author whether an object should be allocated as long-lived or short-lived. These two object types are placed at opposite end of the heap and grow toward the middle. 
Critically, the accounting the `gc` performs is unchanged i.e. the ONLY difference is at which end of the heap an object is allocated; `gc.collect()` still sweeps entire heap.

dpgeorge, 2018-05-11:
> Could the temp-heap be located in the already allocated heap? Couldn't you do some extra accounting to keep track of temp-heap objects so the function "destructor" knows which blocks to free?

This is an interesting idea, but it loses the guarantee that you'll always be able to allocate and that coming back to this same point in the code will yield identical results (wrt to allocation).  Because if you rely on the heap again to allocate then it could be full/fragmented.

peterhinch, 2018-05-11:
The thing that bugs me is how you get anything back from a scoped function, especially when it's an ISR where the return value is lost. It seems similar to the inline assembler where, to return multiple values, you pass mutable objects (in the asm case arrays).

Clearly that won't work in the general case of locally allocated objects for the reason you pointed out above. Could you update a global, bound or arg object containing small ints?

Since floats need allocation I can see that the same trick wouldn't work for them. Yet asm can update an array of floats because they are stored in consecutive machine words. Would updating an external float array work? Presumably MicroPython transcribes an allocated float into a single machine word when it updates an array element.

If it were capable of handling multiple (small) ints and floats alone I can see applications both in ISR's and in maths-intensive routines.

adritium, 2018-05-11:
@dpgeorge 

>  Because if you rely on the heap again to allocate then it could be full/fragmented.

Why can't we have both methods which can each be enabled/disabled separately by a mpconfig.h macro?

adritium, 2018-05-11:
@peterhinch 

> The thing that bugs me is how you get anything back from a scoped function, especially when it's an ISR where the return value is lost.

You'd have to ping-pong between contexts.

```Python
long_term_msgs = []
isDone = False
while isDone is False:
    m1 = busObj1.AllocateAndReturn()
    @micropython.temp:
        m2 = busObj2.AllocateAndReturn()
	@micropython.normal:
	    isDone = IsLoopDone(m2)    
    long_term_msgs.append(m1)
```
`m1` is the value we want to save "long-term" and `m2` is what we only need temporarily.
We have to nest `@micropython.normal` within the context of the `*.temp` otherwise `m2` will get cleaned up and disappear.
It's valid to use `m2` inside a nested `@*.normal` because `@*.*` merely toggles some function pointers controlling allocation on the heap.

**That breaks CPython compatibility!!!!!**. 
Y'all are the syntax experts so here's the above example refactored to use functions.

```Python
long_term_msgs = []
isDone = False
while isDone is False:
    m1 = busObj1.AllocateAndReturn()
    isDone = m2()    
    long_term_msgs.append(m1)
	
@micropython.temp
def m2():
    m2 = busObj2.AllocateAndReturn()
    return getDone(m2)
	
@micropython.normal
def getDone(m2):
    return IsLoopDone(m2)

```

adritium, 2018-05-11:
Another idea: use the same-ish syntax as `nlr_push()/pop`.

It's not ideal because there's no automatic context cleanup but there's also no need to create a function for each nested context.

```Python
long_term_msgs = []
isDone = False
while isDone is False:
    m1 = busObj1.AllocateAndReturn()
	
    micropython.temp()
	m2 = busObj2.AllocateAndReturn()
	micropython.normal()
	    isDone = IsLoopDone(m2)
	micropython.unnormal()
	micropython.untemp()
	
    long_term_msgs.append(m1)
```

peterhinch, 2018-05-12:
Your refactored version fixes the syntax issue, but the problem with switching contexts is that you're allocating on the normal heap within a scoped function. This precludes using the scoped function as an ISR. It also loses the deterministic execution property; arguably this is a minor issue since the context switch could be done once immediately prior to return (as in your example).

@dpgeorge I think this has a lot of potential, both for ISR's and for methods like [this sensor fusion code](https://github.com/micropython-IMU/micropython-fusion/blob/fe72a6870357c8f9b1cae78b98564412382943d9/fusion.py#L116) which does a lot of floating point maths, ultimately returning three floats. This is used in realtime systems so fast deterministic execution is a plus. If only we could figure out how to get the three floats back...

adritium, 2018-06-04:
@dpgeorge 

> there is no GC'ing of this temporary heap

Why? There's nothing stopping `gc` from working.

adritium, 2018-06-04:
Also, requiring heap size to be specified in units of bytes is a bad idea: @micropython.heap_on_stack(200). Py-author and C-module author won't always be the same; how is a user supposed to know

1. How much heap `ObjType1` requires?
2. How many background allocations a given method does?

Python is supposed to abstract away implementation details yet you're demanding your `.py`-user know them which makes a `.py` written with fixed-size scoped-alloc very brittle: they'll be choosing pretty much random values until they finally find the one that's just "big enough" . . . until some implementation changes and breaks their script. This doesn't sound right. Maybe if you want a super-optimized implementation, they should have a choice to use knowledge of implementation details but there should also be a way to auto-configure it, even if it's less optimized.

dpgeorge, 2018-06-05:
> there is no GC'ing of this temporary heap
> 
> Why? There's nothing stopping gc from working.

It's to keep things simple and deterministic, so that you can be guaranteed that it runs the same every time.