micropython/micropython #3777: STM32H7xx UART problems

gradoj, 2018-05-09:
It appears there is the beginning of a port for the STM32H7xx microcontroller running on a Nucleo board. I have programmed a board and verified UART1 works fine attached to the REPL but cannot get UART5 working. 

It appears there is a copy/paste error in uart.c where USART16ClockSelection is being set instead of USART234578ClockSelection :

```
void uart_init0(void) {
    #if defined(STM32H7)
    RCC_PeriphCLKInitTypeDef RCC_PeriphClkInit = {0};
    // Configure USART1/6 clock source
    RCC_PeriphClkInit.PeriphClockSelection = RCC_PERIPHCLK_USART16;
    RCC_PeriphClkInit.Usart16ClockSelection = RCC_USART16CLKSOURCE_D2PCLK2;
    if (HAL_RCCEx_PeriphCLKConfig(&RCC_PeriphClkInit) != HAL_OK) {
        __fatal_error("HAL_RCCEx_PeriphCLKConfig");
    }

    // Configure USART2/3/4/5/7/8 clock source
    RCC_PeriphClkInit.PeriphClockSelection = RCC_PERIPHCLK_USART234578;
    RCC_PeriphClkInit.Usart16ClockSelection = RCC_USART234578CLKSOURCE_D2PCLK1; 
```

I've updated this line (`RCC_PeriphClkInit.Usart234578ClockSelection = RCC_USART234578CLKSOURCE_D2PCLK1;` )but micropython still hangs when uart5.read() is called. I've noticed that when uart5.read() is called the uart_isr_handler is run continuously and never stops. I assume the line of code to read the byte is supposed to clear the interrupt flag but not really sure.

 ```
               // only read data if room in buf
                #if defined(STM32F7) || defined(STM32L4) || defined(STM32H7)
                int data = self->uart.Instance->RDR; // clears UART_FLAG_RXNE
                #else
                int data = self->uart.Instance->DR; // clears UART_FLAG_RXNE
                #endif
                data &= self->char_mask;
                // Handle interrupt coming in on a UART REPL
                if (self->attached_to_repl && data == mp_interrupt_char) {
                    pendsv_kbd_intr();
                    return;
                }
```

Any help appreciated. Thanks.

dpgeorge, 2018-05-10:
> It appears there is the beginning of a port for the STM32H7xx microcontroller running on a Nucleo board.

Yes, but it's still very much a work-in-progress and not everything is expected to work at this point.

> It appears there is a copy/paste error in uart.c where USART16ClockSelection is being set instead of USART234578ClockSelection :

Yes it looks like there is an error here, thanks for pointing it out.

> but micropython still hangs when uart5.read() is called. I've noticed that when uart5.read() is called the uart_isr_handler is run continuously and never stops.

uart_irq_handler will run continuously if the IRQ is not cleared, but that behaviour/bug should be independent of calling uart5.read().  Reading the RDR register should (according to the datasheet) clear the RXNE flag.  Perhaps there is another IRQ that is still pending that is not getting cleared?  But still, it's strange that it works for USART1 but not for UART5.

gradoj, 2018-06-05:
It appears it was legitimate data coming in from a device on UART5 which was overflowing the buffer.  I don't know exactly how best to fix this but it highlights a couple of issues with the UART code when you have a device streaming data:

-If the buffer ever overflows the UART interrupt is disabled and never gets re-enabled. There is code to re-enable but doesn't seem to get run
-The uart timeout never occurs if data is continuously received. This seems to make sense unless you call read() which reads everything available and in my case never returns because it keeps getting data and timeout must keep getting reset.
- Some other problem with the disabling of the UART interrupt causes  `if (__HAL_UART_GET_FLAG(&self->uart, UART_FLAG_RXNE) != RESET) {`  statement to fail in the ISR. Seems to work more consistently if I use `if (__HAL_UART_GET_IT_SOURCE(&self->uart, UART_IT_RXNE) != RESET) {`

dpgeorge, 2022-03-30:
H7 support has been improved a lot recently.  In particular UART clock config was fixed in a708848b0cb0f304f43178140d84d5535e186a38

closed 2022-03-30 by dpgeorge