micropython/micropython #3786: gc_alloc() - should collected=1 after gc_collect() is executed?

adritium, 2018-05-16:
https://github.com/micropython/micropython/blob/1b7487e519c2e2a66f2f5a45ebd238d0773ca3d3/py/gc.c#L452-L458

If a `gc_collect()` is performed here then why isn't `collected` set to `1` just like it is in the next code block?

https://github.com/micropython/micropython/blob/1b7487e519c2e2a66f2f5a45ebd238d0773ca3d3/py/gc.c#L471-L479

Under existing code, it seems you could

1. Perform a `gc_collect()` (from L455)
2. Attempt to locate a large enough block
3. If large enough block not found in **step 2**, perform another `gc_collect()` _even though no allocation operation was performed between **step 2** and **step 3**_
4. Set `collected = 1`
5. Look again for a large enough block as in step 2 - since it failed the first time, it will fail again because memory layout has not change since the first time `gc_collect()` was called on L455
6. If large enough block not found, `return NULL`

By adding `collected = 1` at L455, we'd be avoiding an extra `gc_collect()` + search.

dpgeorge, 2018-05-17:
> By adding collected = 1 at L455, we'd be avoiding an extra gc_collect() + search.

Yes, that looks to be a correct conclusion.

dpgeorge, 2018-05-21:
Fixed by 6bd78741c13849da0570710ad6df80846df812c0

closed 2018-05-21 by dpgeorge