micropython/micropython #3789: MicroPython assumes valid prelude and bytecode in .mpy files

tomlogic, 2018-05-17:
We're running MicroPython as a task in an embedded product, and feel it's relatively safe to run Python code in this sandboxed environment.  It accesses the file system in a restricted context, and Python code shouldn't be able to access data outside of the MicroPython heap and data structures embedded in the firmware.

We're looking at supporting `.mpy` files in this environment.  It seems safe to allow `.mpy` files created on the device and stored such that the user cannot modify the contents.  We'd also like to support use of `mpy-cross` to compile files that require a larger heap.

But we're concerned that users could modify the `.mpy` file in ways that would (for example) allow for reading any memory address or overwriting areas of RAM outside of the MicroPython heap or its task's stack.

I've been looking at file contents outside of the actual bytecode to begin with, and would like to implement some sanity checks on some values.  For example, `n_def_pos_args` must be <= `n_pos_args`.  And it looks like `n_state` should be at least `n_pos_args + n_kwonly_args + 1`.

Is it possible to calculate a value for `n_exc_stack` by doing a validation pass on the bytecode?  Or even a sanity check on the three `_args` settings (ensure the bytecode doesn't reference an arg index beyond what's configured)?  Are there other checks we could perform?

I feel that it's better to add this burden to the import phase and reject invalid `.mpy` files instead of adding range checks to the vm.

We plan to implement these behind a `MICROPY_` configuration macro and eventually submit a PR.  Open to recommendations on a name for that macro.

dpgeorge, 2018-05-17:
> Python code shouldn't be able to access data outside of the MicroPython heap and data structures embedded in the firmware.

There are some ways to access arbitrary memory, eg: `machine.memXX[..]`, `uctypes` module, `array` and `struct` with P, O and S typecodes.  Do you disable these?

> Is it possible to calculate a value for n_exc_stack by doing a validation pass on the bytecode? Or even a sanity check on the three _args settings (ensure the bytecode doesn't reference an arg index beyond what's configured)? Are there other checks we could perform?

Yes probably it will require a validation pass of the bytecode to check the exception stack, the normal value stack, the number of fast locals, and the maximum const obj that is loaded.  A single pass should be enough to verify these things.

More tricky and just as necessary would be to check that jump labels are within range and point to a valid start-of-opcode location.

> I feel that it's better to add this burden to the import phase and reject invalid .mpy files instead of adding range checks to the vm.

The checks would be much easier to add to the VM but for performance reasons I agree it's better to add the checks to the loading of the .mpy file.

> Open to recommendations on a name for that macro.

Maybe `MICROPY_PERSISTENT_CODE_VERIFY`?

jepler, 2018-06-07:
Directing the fuzzer at the execution of mpy files yields up mostly assertion failures, but at least one "real crash" in `mp_execute_bytecode`.

Here's a zip, un-triaged, of crashes and assertion failures in the relevant code:
[mpy-crashes.zip](https://github.com/micropython/micropython/files/2078793/mpy-crashes.zip)

I'm working with a modified version of #3798 changed so that the commandline argument to `micropython_fuzzing` is assumed to be a .mpy file instead of a .py file.  (I also disabled "ubsan" because for whatever reason it was finding a lot more ubsan-type errors which were diluting the result)

Later, for the first 9 crashes I reproduced them by copying the file to `crash.py` and running (not instrumented for fuzzing, but with asserts enabled!) `./micropython -c 'import crash'`.  All 9 crashes reproduced under these circumstances.

`id:000009*` is the first actually interesting one, and it's at that point that I stopped checking that they all reproduced.  It crashes here:

```
Program received signal SIGSEGV, Segmentation fault.
0x00000000004a9af0 in mp_execute_bytecode (code_state=<optimized out>, 
    inject_exc=<optimized out>) at ../../py/vm.c:871
871	                    DECODE_PTR;
```

The first 9 crashes seemed to be at just 5 source locations, so the total number of truly distinct crashes is probably far short of the 434 distinct inputs that afl-fuzz counted as distinct.
```
py/qstr.c:188: qstr qstr_from_strn(const char *, size_t): Assertion `len < (1 << (8 * MICROPY_QSTR_BYTES_IN_LEN))' failed.
py/persistentcode.c:148: mp_obj_t load_obj(mp_reader_t *): Assertion `obj_type == 'f' || obj_type == 'c'' failed.
py/objtype.c:1123: void super_attr(mp_obj_t, qstr, mp_obj_t *): Assertion `MP_OBJ_IS_TYPE(self->type, &mp_type_type)' failed.
py/vm.c:767: mp_vm_return_kind_t mp_execute_bytecode(mp_code_state_t *, volatile mp_obj_t): Assertion `exc_sp >= exc_stack' failed.
```

I would be happy to give more information about the fuzzing setup, or just give the fuzzer another run when some initial improvements to mpy parsing and validation have gone in.

.. and I just noticed I was 45 commits behind micropython's master branch when I tested, so it's entirely possible that some of the detected crashing cases aren't currently crashing.

dpgeorge, 2018-06-09:
Thanks @jepler for looking into this.  At the moment I don't think it's worth trying to fuzz input .mpy files because there is really no protection against them being malformed.  As such we can expect a lot of failures.

For example the `DECODE_PTR` failure in vm.c will not be fixed at the VM level because it would be too expensive.  Instead, loading of .mpy files should verify the bytecode, as discussed above.  After such thing is added then it would be great to try fuzzing it to prove that it works as intended.