micropython/micropython #3790: ports/stm32/rtc.c Setting RTC subsecs has no effect

peterhinch, 2018-05-17:
When setting the RTC time the `datetime` method appears to ignore the value of the `subseconds` arg:
```python
import pyb
rtc = pyb.RTC()

y, m, d, weekday, hrs, mins, secs, subsecs = rtc.datetime()
while subsecs < 192:
    y, m, d, weekday, hrs, mins, secs, subsecs = rtc.datetime()

print(rtc.datetime())
# Printed (2018, 5, 17, 4, 12, 2, 44, 255) which is OK
rtc.datetime((y, m, d, weekday, hrs, mins, secs, 64))
print(rtc.datetime())
# Printed (2018, 5, 17, 4, 12, 2, 44, 255) (??)
```
In my tests after setting the time, regardless of the `subseconds` value passed, `subseconds` always initially reads 255.
```
MicroPython v1.9.3-547-g3d5d76f on 2018-04-11; PYBv1.1 with STM32F405RG
```

dpgeorge, 2018-05-18:
According to the datasheet, the sub-second register in the RTC is read-only.  So there's no way at all to set it.  It seems to be reset (to 255) when you initialise the time.  That's all that can be done.

What is the use-case for setting it?

peterhinch, 2018-05-18:
Ah, I should have looked at the datasheet. The C code seems to be trying to set subsecs [here](https://github.com/micropython/micropython/blob/3e6ab82179f7b9a1df915f4bd0d6e48a454d942c/ports/stm32/rtc.c#L532) so I assumed the hardware was capable.

I'm writing a driver for GPS modules capable of calibrating and setting the RTC and was puzzling over the lack of subsecond accuracy. If the register is read-only that settles the matter :)

closed 2018-05-18 by peterhinch

dpgeorge, 2018-05-18:
> The C code seems to be trying to set subsecs here so I assumed the hardware was capable.

Yes, probably that line should be removed so it's not confusing.

> I'm writing a driver for GPS modules capable of calibrating and setting the RTC and was puzzling over the lack of subsecond accuracy.

Cool!  On a related topic, I'm currently designing a 64-bit, absolute, nanosecond counter based on the RTC + SysTick.  I should be able to provide a reasonably accurate `ticks_ns()` as well as an absolute nanosecond timestamp.  It would be useful to calibrate that against GPS, and maybe there needs to be a way to set the sub-second value.  Currently you can only set the seconds/min/hour/day/month/year, like with the RTC.

peterhinch, 2018-05-19:
You probably know this but popular GPS modules based on the MTK3339 chip provide a 1Hz pulse whose leading edge marks a 1 second boundary to an absolute accuracy of +-10ns. (!)

I'm intrigued by the notion of a nanosecond counter and how you'd use it in a MicroPython context. Or is it aimed at assembler applications?

dpgeorge, 2018-05-21:
Setting the subseconds was removed in 6c955932f3c4bfa56336cbacad389d9f874cba10

> provide a 1Hz pulse whose leading edge marks a 1 second boundary to an absolute accuracy of +-10ns.

Thanks, that's useful to know.

> I'm intrigued by the notion of a nanosecond counter and how you'd use it in a MicroPython context.

With fast CPU frequencies on newer MCUs (eg 216MHz, even 400MHz), microseconds are no longer small enough to get good measurements (timestamps or delta times) on certain things.  It's not so much the need for 1ns resolution but rather sub-microsecond, like 100ns.