micropython/micropython #3792: garbage collection - does it have to be monolithic/atomic?

adritium, 2018-05-17:
A big issue with garbage collection is that it takes so long **and** it's monolithic so when it runs, it must finish before releasing control back to the VM for the next bytecode. 

It would be nice if the user had the option to execute `gc` in chunks so that instead of one big timing hit, you could spread out the overhead. Each time `gc.collect()` ran, it would take a small time slice (while taking more calls to actually finish the job).

To accomplish this, we need to refactor the code to allow for smaller chunks to be executed each `gc` call.
The question is: which of the operations that make up the `gc` must be atomic with respect to the other memory operations that can be performed by the user: allocation and explicitly calling `free`.

peterhinch, 2018-05-18:
A practical solution which has worked for me is periodically to issue **gc.collect()**, ideally when the application is idle. On the Pyboard this can get GC times down to about 1ms.

adritium, 2018-05-22:
I posted this question at [stackoverflow](https://stackoverflow.com/questions/50439859/garbage-collection-does-marksweep-have-to-be-monolithic-atomic) and it turns out this feature [has existed for a long for Java](https://stackoverflow.com/a/50455389/4784683).

Now that I know it's possible to implement, the question is how to do it for uPy.

adritium, 2018-05-22:
@dpgeorge could this be implemented eventually?

dpgeorge, 2018-05-23:
> Without threading, you'd have to execute the gc pieces - the split up functions - using the macros DISPATCH() or MICROPY_VM_HOOK_LOOP, effectively "in-between" the bytecodes.

A good way to do it is just do bits of the GC every time an allocation (alloc, realloc or free) happens.  That way you don't incur any overhead if you don't use memory allocation at all.

> could this be implemented eventually?

It could be, yes.  But I don't have any plans to work on it in the near future, so feel free to look into it yourself.  I'm more interested in providing a complete solution that is 100% deterministic, along the lines of #3774.

adritium, 2018-05-23:
> A good way to do it is just do bits of the GC every time an allocation (alloc, realloc or free) happens. That way you don't incur any overhead if you don't use memory allocation at all.

This feature - call it **gc-in-pieces**  aka **GIP** - would have to be explicitly enabled just like `#define MICROPY_ENABLE_GC`

If enabled, frequency of **GIP** calls would be a configurable parameter to control whether it runs after every byte-code `DISPATCH()`, on a jump `DISPATCH_WITH_PEND_EXC_CHECK()` or only during allocation (alloc, realloc, free). 

_A "**GIP** cycle" is the complete sequence of individual operations that must be finished to equal a call to the monolithic version of `gc`._

### Algorithm to control "number of calls to complete"
The frequency at which **GIP** is called - bytecode, jump or allocation - is separate from the parameter "how many calls does it take to complete a **GIP** cycle". While the first can be set at compile-time, the second should be a function of the runtime profile of the application, specifically how fast the heap is performed. 

If an application consumes the heap very slowly (relative to the frequency of **GIP** calls) then each **GIP** call can afford to do a smaller portion of the `gc` operations resulting in a longer **GIP** cycle which results in less overhead everytime **GIP** is executed.
 
But if an application consumes the heap very quickly . . . then each **GIP** call should do more work/**GIP**.

The goal of the algorithm is to choose as big a `gc` cycle as possible under the constraint of minimizing the number of full `gc`s performed during allocation.

peterhinch, 2018-05-24:
Where applications fail due to allocation issues it is usually a consequence of heap fragmentation: there is ample space for an object, but insufficient contiguous space. My gut feeling is that **GIP** would not be as effective in reducing fragmentation as a periodic issuing of **gc.collect()**.

Unfortunately I can't see how to establish whether that gut feeling is actually correct. What's your view on this?

adritium, 2018-05-24:
But that’s exactly what GIP is: periodic execution in the background executed after every bytecode `DISPATCH()`, jump `VM_HOOK_LOOP` or on an allocation (alloc, reallocate, free). 

The only difference is that the execution of `gc` would be spread out over time so you wouldn’t have one big pause in your script when `gc` is executed.

adritium, 2018-05-28:
I earlier stated

> You only need to finish GIP cycle if you attempt to allocate and you cannot find space.

This is true but incomplete. I should've added **and if the heap has been impacted such that more memory has become available in the middle of the GIP cycle**

But how can an implementation of the **GIP** know whether new memory has become available?

Let's list all the operations that can impact the heap

1. Allocation - only consumes memory. When `gc_alloc` is called, there is no way that memory can be freed.
2. Reallocation - if fewer blocks are requested, memory is freed. If more blocks are requested, technically memory is consumed but there is still a `gc_free` called so to be safe, we should consider that memory was technically freed.
3. `mp_store_name` - the first line consumes memory from the heap. The second line both consumes **and** _implicitly frees_. After 2nd is executed, the object returned by first line becomes garbage because the reference to it is lost i.e. the `.py` script cannot reach it anymore. So, whenever `mp_store_name` is called **and** the identifier exists but something new is assigned to it, we need to consider the heap to be "dirty" and in need of garbage collection.

```Python
var1 = getNewObj(1)
var1 = getNewObj(2)
```
4. `del` - calling delete _maybe_ dirties the heap. 

Memory is allocated, assigned to a identifier - and only to that identifier - so when `var1` is `del`eted, heap becomes dirty. `del` deletes the identifier from the dictionary but it doesn't perform a garbage collection on the object held by `var1`.

```Python
var1 = getNewObj(1)
del var1
```
After line 2, two identifiers point to the object so when `var1` is `del`eted, heap is NOT dirty.
```Python
var1 = getNewObj(1)
myList = [var1]
del var1
```

Without massively modifying the `gc` architecture, it's impossible to tell when `del` truly creates an orphaned memory-block so we have to assume it always does.

adritium, 2018-05-28:
I think it's possible.

To detect if any other bytecode causes a dirtying of the heap, you'd have to perform a full `gc.collect()` after every single bytecode **and** detect whether any orphan block was cleaned up by the sweeper.

The first is accomplished by inserting `gc_collect()` inside **vm.c** `DISPATCH()` macro.
The second by placing a flag inside the `if(free_tail)` to detect when memory was freed.

https://github.com/micropython/micropython/blob/dfeaea144168c3ec3b4ec513cfb201ce93f99f5e/py/gc.c#L300-L305

But I'm not sure if that is exhaustive?
I'm a Python novice but is it possible that the ordering of the bytecodes might, in theory, influence whether the heap is dirtied?

@dpgeorge might have to comment.

If we can get a truly exhaustive list, this feature will be a nice improvement to the `gc`.

dpgeorge, 2018-05-28:
> If this list is exhaustive,

Unfortunately it's not.  For example `list.pop()` will remove an element from a list, potentially making it unreachable and hence collectable.  So could `list.remove`, list assignment, list slice assignment, `dict.pop`, dict assignment (to existing key), lots of others, and potentially extension functions in ports like `uos.dupterm`.

adritium, 2018-05-28:
# Running summary
## Configuration

1. Background - non deterministic, driven by algorithm, minimal/none user config required
2. Manual - deterministic because algorithm is disabled, `.py` author has full control when **GIP** is executed (like when using `gc.collect()`)
3. Mixed - background is on but `.py` author can also choose to manually advance the GIP cycle when most advantageous (like when application is idle). This is like a .py-author provided HINT to the gc: using the context of the .py, the GIP can be run when most advantageous (similar to the way a hint is provided in #3774 ).
For time-sensitive portions of code, background **GIP** can be disabled during runtime (which is different from background GIP being enabled at all using `mpconfig.h` macro).

## Algorithm
### Output
1. **length_gip_cycle** - how many cycles to advance **GIP** next time it's called.  _This definition may look different than the one above but it's just a refactoring of the original one above_

### Input(s)
1. **num_remaining_GIP_calls** - if/when the code reaches **must-finish-point**, how many cycles remain to finish the GIP cycle. This number can be negative which would indicate that the GIP cycle finished early. 

The goal of the algorithm is to choose **length_gip_cycle** such that **num_remaining_GIP_calls** - (-Inf, +Inf) - is as close to ZERO as possible. This is basically a feedback-control system. 

To make the algorithm smarter i.e. a better predictor for **length_gip_cycle**, we should provide more inputs: the heap-dirtying events discussed above. However, a "smarter" algorithm means it's more complex which means it takes either more CPU time OR more memory OR both.
 
The approach to designing the algorithm should be carefully chosen. It would be ideal if it were scalable: instead of designing X fundamentally different algorithms for X different "speeds" or "memory sizes" of microcontroller, we should create ONE algorithm whose time/memory complexity is both parametrizable and increases in fine increments (similar to how the user can select to advance the **GIP cycle** in small increments).

adritium, 2018-05-29:
@peterhinch 
> Where applications fail due to allocation issues it is usually a consequence of heap fragmentation

If you want to deal with it head-on - instead of `gc`-ing "fast-enough" - we'd have to implement a [`gc` that compacts](https://docs.oracle.com/javase/8/docs/technotes/guides/vm/G1.html):

> The heap is partitioned into a set of equal-sized heap regions, each a contiguous range of virtual memory. G1 performs a concurrent global marking phase to determine the liveness of objects throughout the heap. After the mark phase completes, G1 knows which regions are mostly empty. It collects in these regions first, which usually yields a large amount of free space. This is why this method of garbage collection is called Garbage-First. As the name suggests, G1 concentrates its collection and compaction activity on the areas of the heap that are likely to be full of reclaimable objects, that is, garbage. G1 uses a pause prediction model to meet a user-defined pause time target and selects the number of regions to collect based on the specified pause time target.

adritium, 2018-05-29:
@dhylands comments?

dhylands, 2018-05-29:
> @dhylands comments?

Overall, it seems like a reasonable technique. I don't expect it will change fragmentation much, since if its tuned properly, it will wind up collecting at the same times that the regular gc does. What does change is that the collection time will be faster because a bunch of work has already been done.

peterhinch, 2018-05-30:
As an application developer I find that (barring coding errors) OOM failures invariably result from fragmentation. There are tricks to avoid it, such as allocating large buffers early and ensuring they never go out of scope; also issuing manual GC in idle periods and after critical operations. It's an aspect you have to bear in mind at the outset when designing any nontrivial application.

Ideally we need a benchmark which could hammer the heap and issue some kind of fragmentation metric so we could compare different GC algorithms. In my view fragmentation is crucial: more important even than performance.

dpgeorge, 2018-05-30:
> In my view fragmentation is crucial: more important even than performance.

Yes, I agree.  No point having performance if there's no memory to perform in :)  That's why I think fragmentation needs to be solved completely, not just alleviated in certain cases (even alleviated in 99% of cases wouldn't be enough).  Of course the crude solution is to just issue a reset periodically, but that's pretty crude.  I think scoped allocation as discussed in https://github.com/micropython/micropython/issues/3774#issuecomment-387963973 is the way to go.

adritium, 2018-05-30:
@dhylands 

> What does change is that the collection time will be faster because a bunch of work has already been done

This is not accurate: **GIP** spreads the work out over time. If `gc` currently occupies X% of time in your application, with this method it will still occupy X% (plus a little more for the overhead of extra function calls).

adritium, 2018-05-30:
@dpgeorge 

> Of course the crude solution is to just issue a reset periodically, but that's pretty crude. I think scoped allocation as discussed in #3774 (comment) is the way to go.

#3774's scoped alllocation allocates a temporary heap (1) which disappears after function exits and (2) [there is no GC'ing of this temporary heap](https://github.com/micropython/micropython/issues/3774#issuecomment-387963973) ; how would that solve fragmentation? 

To solve fragmentation, the collector should compact during collection like [G1 collector](https://docs.oracle.com/javase/8/docs/technotes/guides/vm/G1.html)

adritium, 2018-05-30:
There's a bug my algorithm (in my assumption).

Summary: because we can never safely assume we have an exhaustive list of heap-dirtying events, the input **num_remaining_GIP_calls** is (completely?) irrelevant.

Assume:
* we DO have an exhausitve list of ALL heap-dirtying-events (HDEs)
* we get to the must-finish-point (i.e. initial search for free memory fails in `gc_alloc()`)
* Ignore fragmentation 
If we haven't received a HDE since the start of **GIP** cycle, it suffices to finish the remaining **GIP** cycles before trying to allocate again. If there are no remaining **GIP** cycles to execute, then we can conclude there is no more memory left. This is because if no HDEs occurred in the middle of the gc in pieces **GIP**, there is guaranteed to be no more memory available. Furthermore, if the collector KNOWS about ALL of the HDEs, it can know whether performing a full collection will POSSIBLY yield any new memory. Bonus: with knowledge of ALL HDEs, you would only have to START the **GIP** cycle when an HDE occurs i.e. it wouldn't have to run all the time.

However, [as per @dpgeorge it is not feasible to create a complete database of all HDEs](https://github.com/micropython/micropython/issues/3792#issuecomment-392504259). Therefore, we must assume that a HDE can occur with any frequency and at any time with respect to **GIP** cycle. Consequently, if we get to the **must-finish-point**, it would not suffice to finish the remaining cycles of **GIP** because an unknown HDE could've occurred anywhere during **GIP** cycle; we would have to perform the complete collection monolithically at **must-finish-point** even if there is 1 **GIP** cycle left to finish. This means the value of **num_remaining_GIP_calls** is completely irrelevant as an input to the algorithm. 

Well, I shouldn't say completely irrelevant. The fact that you know SOME of the HDEs could be incorporated into the model: if no known HDEs occur but collection still results in the freeing of memory then over time, you could infer (maybe?) something about the ratio of unknown-to-known HDEs??? Could get complicated but maybe for long-running applications it's worth it but for simpler (and less capable hardware) it's not. 

All the more reason to build this algorithm in layers though maybe we should have at least 2 versions if that's not possible.

adritium, 2018-05-30:
@dhylands 

> I don't expect it will change fragmentation much, since if its tuned properly, it will wind up collecting at the same times that the regular gc does

Can you elaborate why you think that?

At first I thought that because you can allocate and free [in the middle of **GIP** at any point](https://github.com/micropython/micropython/issues/3792#issuecomment-391007214), fragmentation could only decrease.

Specifically, I thought that because blocks would be freed sooner - because sweep-phase is happening "in parallel" to the application - you'd be able to find space sooner . . . but I can't quite convince myself this is true.

dhylands, 2018-05-30:
You can't actually free any blocks until you've done 100% of the sweeping. It could be that the very last block references a block early in the heap.

If you happen to complete your incremental mark/sweep before you run out of memory, then you could do the free step early. The earlier that you can do frees, the more impact it will have on reducing fragmentation. But if you free early then you're also using more CPU cycles than freeing when running out of memory.

I was assuming that "tuned" meant you'd be freeing in the incremental model at approximately the same time as in the non-incremental model, in which case there would be minimal differences in the fragmentation pattern.

The big thing that incremental gc buys you is that you even out the load required to perform a gc, so you get more deterministic execution times (i.e. reduce that big delay that happens in the non-incremental model). If you tune things to free early then you'll get reduced fragmentation at a cost of performing more collection cycles then in the non-incremental model.

The amount of fragmentation is inversely proportional to the number of times that you collect. Which is why adding gc.collect calls into your code at "key" places will help reduce fragmentation.

If you went to the extreme and collected after every opcode then you'd get the least amount of fragmentation possible, but it wouldn't necessarily avoid bad fragmentation. Some allocation patterns cause bad fragmentation.

adritium, 2018-05-30:
> You can't actually free any blocks until you've done 100% of the sweeping. It could be that the very last block references a block early in the heap.

I don’t understand. 

Once you finish the mark-phase, you can sweep at the granularity of a single block: mark it as free and relinquish control. 

> If you happen to complete your incremental mark/sweep before you run out of memory, then you could do the free step early. 

I think we’re using different terminology. In `gc_sweep()` blocks are freed one-by-one. You should be able to execute one run of that loop and relinquish control. 
Completing the mark+sweep IS the same thing as freeing memory.

adritium, 2018-05-30:
@dhylands 

> You can't actually free any blocks until you've done 100% of the sweeping.

I think there's a terminology misunderstanding.

When I say _sweep_ I'm talking about the distinct phase, implemented in `gc_sweep()`, that changes a block's `AT_HEAD` status (and its attached `AT_TAIL` blocks if any) to `AT_FREE`.

The mark-phase, run before sweep, is implemented in `gc_collect_start()->gc_collect_root()` and it merely indicates which blocks are reachable by the code by changing their status to `AT_MARK`.

Neon22, 2018-05-30:
The Symbolics machine used mark and sweep as its primary GC mechanism. Of course it had a tagged architecture (40bits for a 32bit word) so any memory word could be marked as garbage with a bit set (Tags were also used for typing so the ALU microcode could be efficient when performing ops.)

Anyway the consensus (https://en.wikipedia.org/wiki/Garbage_collection_(computer_science)) these days is apparently that we either need 5x the memory of a non GC'd system or we could use Tagged pointers like iOS (amongst others) is using. https://en.wikipedia.org/wiki/Tagged_pointer.
I.e. use some of the address space bits which are unused, as we generally work in low RAM environments, to store this kind of data.

I speak generally because we may already be doing this. I don't know.

dhylands, 2018-05-30:
You are correct. I'm using incorrect terminology (I was using memory rather than looking at the code), and also drew some incorrect conclusions.

So yes, using the incremental model will start freeing objects earlier than the non-incremental model, and because of this, you should experience less fragmentation. How much less will still be dependent on the allocation pattern, and what the relationship is between the time taken to mark versus the time taken to sweep.

I still view incremental collection as mostly being useful for getting more deterministic execution times. I think something like #3586 will have a much more significant effect on fragmentation. I also don't look at this as an either/or situation. Having incremental collection and #3586 would both be advantageous.

adritium, 2018-05-30:
@dhylands why not move to the G1 collector which builds-in compaction?

dhylands, 2018-05-30:
Allowing for compaction either requires you to use a master pointer table and double indirection, or somehow recognizing which memory patterns are pointers and adjusting them after moving the object, or using a more rigid object model. A more rigid object model would also allow the gc to determine exactly where pointers are, but this would require that the objects take up more memory.

The current code will occasionally not free an object because a memory pattern discovered during the mark looks like a pointer, but it might just happen to be an integer with a value that happens to coincide with a pointer into the heap.

G1 says that its appropriate for multiprocessor systems with large memories, which is the opposite end of the spectrum that we're dealing with here.

adritium, 2018-05-30:
As I was editing my comment above, I realized (as you stated) that we cannot implement compaction because the address returned is the physical address.

```C
byte * a = m_new(8)
... // compaction happens and moves the 8-byte block
a[0] = 55; // uh-oh! We might not be pointing to the original 8-byte block
```

Even if we returned a virtual address, all pointer operations would be broken because all pointer operations operate on physical addresses. If we used C++ we could define our own pointer class and overload all the operands . . . but I know that's not gonna happen.

cc @peterhinch

jpochyla, 2018-05-31:
@dhylands I think a possible way to a precise GC would be to separate the heaps to a one that could contain pointers, and one that cannot. Which would be a performance hit for objects such as bytes and str, as they share the allocation now.

adritium, 2018-05-31:
That’s not possible

int *b = new(int)
int * a = new(int)

*a = b 
…
*a = 99* 55

You cannot distinguish between a pointer or int with meta information. And you cannot store metadata without hw support. If you do it in software and use any of the bits to tag it, all code that uses pointers will have to be modified to strip out the metabits before using the pointer.

adritium, 2018-05-31:
@dpgeorge 

> That's why I think fragmentation needs to be solved completely, not just alleviated in certain cases (even alleviated in 99% of cases wouldn't be enough)

Without pointer tagging or a way to overload pointer operators, it's impossible to solve it completely. The best you can do is to mitigate fragmentation by improving the allocator. #3586 does this by categorizing allocations as short- or long-lived but to make optimal use of this scheme, `.py` author should provide hints #3774 to `gc`  and then you have to maintain these hints (if you use them) throughout your code . . .which sounds very messy. If that were the only way, fine but there is another way.

Stackoverflow says that fragmentation can be mitigated by subdividing heap (logically) into areas holding [different-sized allocations](https://stackoverflow.com/a/2276452/4784683).

jpochyla, 2018-05-31:
> You cannot distinguish between a pointer or int without meta information.

I think it is possible. Small ints are already tagged on the highest bit, big-int backing storage would get allocated on the data heap.

adritium, 2018-05-31:
Those are Python objects, not fundamental types. There's nothing stopping a `unsigned long` from looking like a `SMALL_INT`. If you want to separate their storage, you'd have to rely on the user to use two separate malloc APIs: one for `mp_obj_t` and one for fundamental types.

And even an `mp_obj_t` can hold fundamental types; here's `led.c` code.
```C
typedef struct _pyb_led_obj_t {
    mp_obj_base_t base;
    mp_uint_t led_id;
    const pin_obj_t *led_pin;
} pyb_led_obj_t;

STATIC const pyb_led_obj_t pyb_led_obj[] = {
    {{&pyb_led_type}, 1, &MICROPY_HW_LED1},
#if defined(MICROPY_HW_LED2)
    {{&pyb_led_type}, 2, &MICROPY_HW_LED2},
#if defined(MICROPY_HW_LED3)
    {{&pyb_led_type}, 3, &MICROPY_HW_LED3},
#if defined(MICROPY_HW_LED4)
    {{&pyb_led_type}, 4, &MICROPY_HW_LED4},
#endif
#endif
#endif
};

STATIC mp_obj_t led_obj_make_new(const mp_obj_type_t *type, uint n_args, uint n_kw, const mp_obj_t *args) {
    // check arguments
    mp_arg_check_num(n_args, n_kw, 1, 1, false);

    // get led number
    mp_int_t led_id = mp_obj_get_int(args[0]);

    // check led number
    if (!(1 <= led_id && led_id <= NUM_LEDS)) {
        nlr_raise(mp_obj_new_exception_msg_varg(&mp_type_ValueError, "LED %d does not exist", led_id));
    }

    // return static led object
    return (mp_obj_t)&pyb_led_obj[led_id - 1];
}

```

So to make this work, you'd have to:
1. only "ALLOW" `mp_obj_t` types to only contain other `mp_obj_t`.
    There's no way to enforce that and tons of code has to be changed.
2. force user to use one `malloc` for `mp_obj_t` and another for non `mp_obj_t`
    Again, no way to verify this at compile or runtime.

And that still doesn't help you with the "dude! where's my memory" problem i.e. double-indirection

```C
typedef struct _msg_mp_obj_t {
	mp_obj_base_t base;
	mp_obj_t mp_obj_id;
	mp_obj_list_t * mp_list_data;
	mp_obj_int_t * mp_int_timestamp;
} msg_mp_obj_t ;

// mp_new_obj allocates on separate `mp_obj_t` heap area
msg_mp_obj_t  * msg = mp_new_obj(msg_mp_obj_t);
// msg is now equal to 0x1000
// . . . other mp_new_obj() are called causing a compaction . . .
// . . . and the block that was at 0x1000 is now at 0x2000 but wait! . . .
// . . . msg_mp_obj_t  * msg is still 0x1000
// so now when I do this:
msg->mp_list_data = mp_new_obj(mp_obj_list_t);
// I'm not referring to the correct block which is at 0x2000
```

For this to be correct, the compaction would have to change `msg_mp_obj_t  * msg` from 0x1000 to 0x2000. How are you gonna do that? Scan the stack and change all `0x1000` to `0x2000`? 
How do you know if `0x1000` in some memory location is a pointer or just an `mp_int` holding `0x1000`? 
You can't.

adritium, 2018-05-31:
@jpochyla yes let's leave compaction for another issue as it's clear the object-model and/or pointer operations have to be modified in a way that breaks compatibility with current code.

However, `gc` in pieces aka incremental `gc` is merely a refactoring of existing code.

aykevl, 2018-06-01:
So I'm reading a book right now (The Garbage Collection Handbook, very much recommended!) about memory allocation and I'm planning to implement a compacting garbage some time for Go on a microcontroller. I've seen a few comments here which aren't entirely accurate/complete, which I'd like to give some background on.

> This feature - call it gc-in-pieces aka GIP - would have to be explicitly enabled just like #define MICROPY_ENABLE_GC

This is called incremental GC.

> compacting GC

A compacting GC is really difficult to write. You will need to have a fully precise GC for that, meaning the GC knows for every memory location with complete certainty whether it is a pointer or not. The current GC is conservative (not precise), meaning as long as something kinda looks like a pointer it is assumed to be one. This is not a big problem: the only disadvantage is that sometimes memory is retained which is really unreachable.

A compacting GC needs to be precise because memory has to move and all addresses pointing towards it have to be adjusted. This is _really, really difficult_. The Go team (with several very talented developers) needed a few years for this, and they had influence on the compiler which we don't have.

Finding pointers on the heap and in global variables is not easy but not all that difficult either. The big problem is finding pointers on the stack and in local registers. I am aware of two ways to do that:

1. Every pointer is indirect. Before resolving a pointer (read+write), it has to go through some sort of translation layer which manages moving objects. This means a big performance hit, as pointers are used everywhere.
2. The compiler keeps track of all pointers on the stack. C is very uncooperative (almost hostile) to programs that want to do such things. This is the most efficient approach, possibly even faster than the current GC on MCUs (it certainly is faster on systems with a CPU cache). You might be able to do it with DWARF debug info or exception handling code, but I haven't tried this and don't t know whether it is possible. Or you might be able to use LLVM (Clang) bitcode, but that means changing the compiler.

All in all I have my doubts this is actually feasible in MicroPython.

dhylands, 2018-06-01:
squeak (a version of smalltalk)  seems to do it (allow objects to be moved) by having a rigid object structure.

adritium, 2018-06-01:
@aykevl you’re relying on an app to sleep and on explicit timing (seconds); this won’t work. 
My `.py` scripts never call sleep; I run a cooperative RTOS which means that there technically is no “idle” time.

The whole point of incremental GC is that monolithic GC can be split up to arbitrarily small pieces but now you’re introducing a time requirement? Introducing explicit time will complicate things.

peterhinch, 2018-06-01:
> how will you give interrupts access to the heap (even just reading) while it is compacting? Some pointers will be adjusted and some won't.

That kills compacting stone dead. The ability to write low-latency ISR's in Python is crucial in my opinion.

peterhinch, 2018-06-01:
An ISR may need to access a pre-allocated buffer, a **bytearray** operating as a ring buffer being just one example.

adritium, 2018-06-01:
If we add “IsCompactible” bit, we can achieve low-latency ISR and ability to read and write to it.

adritium, 2018-06-01:
@dpgeorge 
[> For sure there needs to be a big change to make the GC precise. And even more change to get to a compacting collector. IMO this can't be done by incremental changes to master, but rather it would be worked on in a separate branch to investigate how feasible it is first, and find out if there are any big issues that have not yet been understood.](https://github.com/micropython/micropython/issues/1161#issuecomment-393858843)

Let's get back to the to the point of this issue: incremental `gc`. Is there anything else to be discussed?
@dhylands @peterhinch @aykevl @Neon22

peterhinch, 2018-06-02:
It's an interesting approach. But I'd like to see an actual use case where it would significantly improve an application's performance compared to the approach of issuing **gc.collect()** in idle periods.

In the case of asynchronous code (which covers many of the more complex applications) an individual task can expect delays of a few ms while other tasks are scheduled. So you have a coroutine running **gc.collect()** in a loop; if **gc.collect()** runs sufficiently frequently it can take as little as 1ms. So (in my opinion) the GC overhead in most asynchronous applications is lost in the noise.

Using this [priority mechanism](https://github.com/peterhinch/micropython-async/blob/master/FASTPOLL.md) it's possible to ensure that garbage collection occurs only when other tasks are waiting on a nonzero delay, reducing the GC overhead to zero.

This isn't intended as a criticism: if it doesn't adversely affect fragmentation there is no downside, and it potentially saves application developers from having to consider GC issues. But I question whether effort might more profitably be directed at heap fragmentation.

adritium, 2018-06-03:
> But I'd like to see an actual use case where it would significantly improve an application's performance compared to the approach of issuing gc.collect() in idle periods.

I never said it would improve performance. I said it would reduce **jitter** by taking the time spent in `gc` and spreading out more evenly. In fact it would make **performance** slightly worse from the overhead of extra function calls to perform `gc` incrementally, depending on how often **GIP** is executed.

You get lower jitter while getting slightly worse performance.

> gc.collect() in idle periods

You're assuming that there **are** idle periods. 

Even @aykevl said 

> I think this will work well in applications that sleep frequently.

Not all applications call sleep. Some are going to be just a few for-loops and even those can still make use of **GIP** to reduce jitter.

> But I question whether effort might more profitably be directed at heap fragmentation.

Heap defragmentation involves moving heap blocks around which needs double-indirection which will break all code that uses pointers. 

@aykevl 
> Every pointer is indirect. Before resolving a pointer (read+write), it has to go through some sort of translation layer which manages moving objects. This means a big performance hit, as pointers are used everywhere.

You'll have to introduce new pointer API.

peterhinch, 2018-06-04:
I haven't encountered an application where GC jitter was an issue. On platforms such as the Pyboard timing critical code can usually be written using hardware timers and interrupts. Do you have any use cases?

My comment on reducing fragmentation was referring to [scoped allocation](https://github.com/micropython/micropython/issues/3774#issuecomment-387963973) and the other ideas under discussion short of a full compacting GC.

adritium, 2018-06-04:
> hardware timers and interrupts

For my platform, we're not going to allow those. We're adding scripting as a secondary functionality that should not affect the currently running application.

> Do you have any use cases?

Yes I do. (Don't get hung up on the fact that `device.CAN.queue` may not behave like a normal Py `queue`)

```Python
def func(q1, q2):
	while True:
		q2.put(q1.get())
		
can1 = device.CAN.CAN1
can1.configure(0, 0, 1000000)
q1 = device.CAN.queue(can1)
q1.enable()

can2 = device.CAN.CAN2
can2.configure(0, 0, 1000000)
q2 = device.CAN.queue(can2)
q2.enable()

func(q1, q2)
```
Eventually, `gc` will be triggered and you'll have a big pause which you'll see it in on the bus as a gap in the data. I know scoped-alloc is prime candidate for this but that makes it less beginner-friendly. With **GIP**, a beginner doesn't need to add "magic code" to get rid of the jitter.

Scoped allocation - with a fixed size - is super inflexible because you cannot `gc` it: you can only use it if the object you allocate disappears outside the scope i.e. it's all or nothing (except if you copy it to an existing object _which has to be the right size which is not possible to guarantee if your object has no fixed size_). _It is most useful inside an interrupt._ 

Also, requiring heap size to be specified is a **bad** idea: `@micropython.heap_on_stack(200)`. Py-author and C-module author won't always be the same; 
1. How is a user supposed to know how much heap `ObjType1` requires? 
2. How is anyone supposed to know how many background allocations a given method does? 

Your `.py`-user will be choosing pretty much random values until they finally find the one that's just "big enough" . . . until the `gc` or `ObjType1` implementation is changed .

Compacting `gc` would allow you to "promote" objects based on longevity when you do a `gc`.

dpgeorge, 2018-06-05:
> In some cases it might be impossible or awkward to preallocate. What if allocation only occurs when you call a specific function - which has a side effect - with an argument of a specific type?

Then you should test for all paths of execution and pick a scope-alloc size which covers the worst case.  In embedded programming this is what you have to do.  If you want to guarantee that code always executes the same and always within a certain time limit then you have to know its worst case behaviour, in both time and memory usage.

adritium, 2018-06-05:
Then there should be a way to query the max heap usage during debut phase otherwise you cannot find out what the worst case is.

adritium, 2018-06-05:
@dpgeorge

You bring up two issues: temp-heap size and preallocated object size. 

If it’s the user’s responsibility to assign a temp-heap size, then there should be a mechanism for the user to obtain the max heap size used during the “tuning” phase. 

A separate issue is copying the temp-object to normal heap. If you’re in an interrupt, you must use a pre-allocated object to copy temp-object into normal-heap object but if an object can grow in size in a way that’s only apparent by viewing the C-code, you’ve got a problem.

dpgeorge, 2018-06-05:
> If it’s the user’s responsibility to assign a temp-heap size, then there should be a mechanism for the user to obtain the max heap size used during any given path of execution.

Yes that makes sense.

> A separate issue is copying the temp-object to normal heap. If you’re in an interrupt, you must use a pre-allocated object to copy temp-object into normal-heap object

If you're in a hard-interrupt handler then allocation on the normal heap is not allowed (it's locked).  So there's no way to "copy out" heap objects from a hard interrupt.  You can instead pass out small ints (eg write to a global variable), or store data into global arrays (float, int, byte, etc).

adritium, 2018-06-05:
@dpgeorge 

> If you're in a hard-interrupt handler then allocation on the normal heap is not allowed (it's locked)

Copying to pre-allocated object doesn't require heap operations: if there's a pre-allocated object of the right size you can deep-copy your temp-heap object out to the normal heap.

stinos, 2018-06-05:
>  you can deep-copy 

That's quite hard though, as discussed earlier in e.g. 3652

adritium, 2018-06-05:
What's hard about it other than deep-copy is not implemented for each object?

stinos, 2018-06-05:
Well, that exactly. Right now it's not implemented at all, and a usable implementation which isn't limited to a small subset of types is going to require quite a lot of changes it seems.

adritium, 2018-06-05:
If you need finalization, you have to implement `__del__` on a per-type basis. Period. Would it be more useful if uPy could magically deduce how to finalise your object? Sure it would but it can't.

If you need deep-copy, you have to implement `__deep_copy__` on a per-type basis. Period. Would it be more useful if uPy could magically deduce how to perform a deep-copy? Sure it would but it can't.

@dpgeorge 

> there's no way to "copy out" heap objects from a hard interrupt

Require that a `__deep_copy__` be implemented for each type you want to deep-copy just like you require that a `__del__` be implemented if you want destructor-like behavior.

adritium, 2018-06-06:
However “hard” it is to implement deepcopy, if you want to copy-out an object from an interruption deepcopy has to be implemented. 

The only alternative is to radically bloat the storage of objects on the heap so you can avoid an effort about which the worst you can say is “it’ll take a really long time to implement and it’s boring”.