micropython/micropython #3796: make test ERRORs

jiapei100, 2018-05-19:
Hi, just wanna know if this is normal? I'm using Ubuntu 18.04 + micropython git of the day May 19, 2018.

> ...
> pass  unix/ffi_float2.py
> 548 tests performed (17957 individual testcases)
> 547 tests passed
> 9 tests skipped: framebuf1 framebuf16 urandom_extra vfs_fat_fileio vfs_fat_fsusermount vfs_fat_oldproto vfs_fat_ramdisk buffered_writer extra_coverage
> 1 tests failed: syntaxerror
> Makefile:190: recipe for target 'test' failed
> make: *** [test] Error 1
> 

Thank you 
Pei

peterhinch, 2018-05-20:
I can't replicate this. With source downloaded this morning I get
```shell
pass  unix/ffi_callback.py
pass  unix/ffi_float.py
pass  unix/ffi_float2.py
656 tests performed (18970 individual testcases)
656 tests passed
23 tests skipped: builtin_help builtin_range_binop class_delattr_setattr namedtuple_asdict sys_getsizeof cmd_parsetree framebuf1 framebuf16 framebuf2 framebuf4 framebuf8 framebuf_subclass urandom_extra vfs_basic vfs_fat_fileio1 vfs_fat_fileio2 vfs_fat_more vfs_fat_oldproto vfs_fat_ramdisk buffered_writer resource_stream schedule extra_coverage
[adminpete@axolotl]: /mnt/qnap2/data/Projects/MicroPython/micropython/tests
$ 
```
The build script I use is
```shell
#! /bin/bash
# Update MicroPython source and prepare for build

cd $MPDIR
echo Working...
git checkout master
git pull origin master
git submodule update --init
cd mpy-cross
make clean
make -j 8
cd ../ports/stm32
make BOARD=PYBV11 clean
make BOARD=PYBV10 clean
make BOARD=PYBLITEV10 clean
cd ../esp8266
make clean
cd ../unix
make clean
make -j 8 axtls
# If you're going to enable deplibs: see micropython/README
#make deplibs
make -j 8
```

robert-hh, 2018-05-20:
I support @pythoncoder's observation. No errors with make test.

dpgeorge, 2018-05-20:
It could be the version of CPython that is used to run the tests against.  @jiapei100 what is the contents of `tests/syntaxerror.py.out` after running the test suite.?