micropython/micropython #3805: esp32: UART(0) is disabled (dedicated to REPL)

dreasGh, 2018-05-21:
Hi @dpgeorge,
in response of to the pull request #3784 relative to the dupterm of the esp8266, my need is to use the UART(0) it's not because the other ports are in use, but because my society production version of the board esp32 use RXD0 and TXD0 to communicate with the device.
By now the esp32 was produced so we can't change the hardwere, is there any way to detach the UART(0) from the REPL using the dupterm function?

I hope i made my point.
Thank you for your reply in advance.

robert-hh, 2018-05-21:
I never did that, but the ESP32 is pretty flexible in mapping pins to functions. I may be possible to map UART0 to other pins and UART1 to the pins used by UART0 before.

dreasGh, 2018-05-21:
I made a test changing the pin settings as follow:
`UART(0, baudrate=115200, bits=8, parity=0, stop=1, timeout=0, timeout_char=1, tx=35, rx=34)`

but i'm getting this error message:
```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: UART(0) is disabled (dedicated to REPL)
```

later i've tested with the port 1:
`UART(1, baudrate=115200, bits=8, parity=0, stop=1, timeout=0, timeout_char=1, tx=35, rx=34)`

but I get:
`E (859244) uart: uart_set_pin(536): tx_io_num error`

robert-hh, 2018-05-21:
Pin 35 is input only. Therefore it fails. But from Webrepl I can write:
```
from machine import UART
uart=UART(1, tx=1, rx=3)
```
From that point on I can read & write uart at the pins of UART0.

dpgeorge, 2018-05-22:
> my society production version of the board esp32 use RXD0 and TXD0 to communicate with the device

What is it exactly that you want to do: do you want a UART REPL?  If so on which pins?

dreasGh, 2018-05-22:
Hi @robert-hh,

thanks for help me. I solved my problem using these pins. I used pins number 34-35 because the datasheet of the esp32 describes the RXD0 and the TXD0 with these pins, but i didn't realize that they are mapped like GPIO 1 and 3.

@dpgeorge we are developing an irrigation control unit that use the esp32 like a bridge to a mobile app which allow the user to remote control the unit.
Just like i wrote you in the last comment, the production boards have been produced yet and we used the pin RXD0 and TXD0 for the communication with the control unit.

Anyway i solved the problem.
Thanks for the help.

closed 2018-05-22 by dreasGh

leozlee, 2019-07-11:
@dreasGh Hallo, I have the same problem with u, Could please tell me how to deal with it with more detail?

QGB, 2022-05-18:
```
Continuing.
Note: automatically using hardware breakpoints for read-only addresses.
[New Thread 1070149892]
[Switching to Thread 1070149892]

Thread 4 hit Breakpoint 2, loop () at C:\test\ESP32\hello_world\main\hello_world_main/hello_world_main.ino:20
(gdb) n
(gdb) s
printf (fmt=fmt@entry=0x3c020120 "printf loop %i.\n")
    at /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/riscv32-esp-elf/src/newlib/newlib/libc/stdio/print
f.c:52
/builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/riscv32-esp-elf/src/newlib/newlib/libc/stdio/printf.c: No
 such file or directory.
(gdb)
```