micropython/micropython #3806: does WiPy 2.0 support?

pedrofelipest, 2018-05-21:
Is it supported via SPI communication with nfr24l01?

dpgeorge, 2018-05-22:
WiPy 2.0 is not directly supported by this code repository.  You might be able to use the generic esp32 port from this repository, and that has working SPI.

robert-hh, 2018-05-22:
I have no WiPy 2 here, otherwise I could test it. It should work. The only thing to consider is the WiFi antenna switch on GPIO16 of the WiPy2, where you can switch between internal and external antenna.

robert-hh, 2018-05-22:
OK. So it's the opposite. You use WiPy2 software on a generic ESP32. That works for devices with a Rev0 chip. It should be possible to adapt the driver to the WiPy2 firmware. But this is the wrong place to ask. Actually, no place really fits. If you have concrete error messages, you may ask in the Pycom Forum at https://forum.pycom.io/.  But just tell that you are using a WiPy2. Or ask in the micropython forum. But @pythoncoder, who provided the driver and is very supportive, is not a user of the Pycom firmware branch, AFAIK. 
Maybe at some point Bluetooth will be better usable in this port.