micropython/micropython #3807: Hardfault when mp_parse is interrupted with pendsv.

iabdalkader, 2018-05-21:
Hello, I am getting a hardfault (INVSTATE) in nlrthumb.c if mp_parse is interrupted with pendsv. Is this behavior expected ? Could some shed more light on the cause ?

Here's some info from GDB:
```
Breakpoint 1, HardFault_C_Handler (regs=0x2001fcc8) at stm32_it.c:144
144	    if (!pyb_hard_fault_debug) {
(gdb) backtrace 
#0  HardFault_C_Handler (regs=0x2001fcc8) at stm32_it.c:144
#1  <signal handler called>
#2  nlr_jump (val=0x200117e0) at ../../py/nlrthumb.c:85
#3  0x08092dd4 in mp_parse (lex=0x20012140, input_kind=(MP_PARSE_FILE_INPUT | unknown: 134819284)) at ../../py/parse.c:925
#4  0x0000004e in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb) print /x *regs
$1 = {r0 = 0x200117e0, r1 = 0x20, r2 = 0x4, r3 = 0x4, r12 = 0x2000, lr = 0x8092dd5, pc = 0x808ff84, xpsr = 0x100f000}
(gdb) print /x *SCB
$3 = {CPUID = 0x411fc270, ICSR = 0x400803, VTOR = 0x8020000, AIRCR = 0xfa050300, SCR = 0x0, CCR = 0x70200, SHPR = {0x0, 0x0, 0x0, 0x0, 0x0, 
    0x0, 0x0, 0x0, 0x0, 0x0, 0xf0, 0x0}, SHCSR = 0x0, CFSR = 0x20000, HFSR = 0x40000000, DFSR = 0xb, MMFAR = 0x0, BFAR = 0x0, AFSR = 0x0, 
  ID_PFR = {0x30, 0x200}, ID_DFR = 0x100000, ID_AFR = 0x0, ID_MFR = {0x110030, 0x0, 0x1000000, 0x0}, ID_ISAR = {0x1101110, 0x2112000, 
    0x20232231, 0x1111131, 0x1310132}, RESERVED0 = {0x0}, CLIDR = 0x9000003, CTR = 0x8303c003, CCSIDR = 0xf00fe019, CSSELR = 0x0, 
  CPACR = 0xf00000, RESERVED3 = {0x0, 0x800, 0x0 <repeats 23 times>, 0x30003, 0x0, 0x0, 0x1000001, 0x0 <repeats 64 times>}, STIR = 0x0, 
  RESERVED4 = {0x0 <repeats 12 times>, 0xc0000000, 0xfffffff8, 0x0}, MVFR0 = 0x10110221, MVFR1 = 0x12000011, MVFR2 = 0x40, RESERVED5 = {0x0}, 
  ICIALLU = 0x0, RESERVED6 = {0x0}, ICIMVAU = 0x0, DCIMVAC = 0x0, DCISW = 0x0, DCCMVAU = 0x0, DCCMVAC = 0x0, DCCSW = 0x0, DCCIMVAC = 0x0, 
  DCCISW = 0x0, RESERVED7 = {0x0, 0x0, 0x0, 0x0, 0x0, 0x0}, ITCMCR = 0x73, DTCMCR = 0x43, AHBPCR = 0x9, CACR = 0x2, AHBSCR = 0x800, 
  RESERVED8 = {0x0}, ABFSR = 0x0}
```

RockySong, 2018-05-22:
Awesome! I also met several kinds of hardfault when porting openMV to NXP i.MX RT1050, and can recover INVSTATE type hardfault by clearing bit 15,14 of stacked XPSR. In my case, INVSTATE type hardfault always caused by wrongly set of bit 15, 14 of stacked XPSR in stack, when returing from pendSV handler to nlr_jump().
Does this only cause INVSTATE type hardfault, or it can also cause other types of hardfault (I met precise, imprecise, unaligned types hardfault too).

dpgeorge, 2018-05-22:
> Hello, I am getting a hardfault (INVSTATE) in nlrthumb.c if mp_parse is interrupted with pendsv.

Are you sure it's going through pendsv, or is it just simply calling nlr_jump from mp_parse?  What triggered this hard fault?  What MCU is it using?

iabdalkader, 2018-05-22:
Hi, no I'm explicitly setting pendsv from USB interrupts to stop/exec scripts:

```
    MP_STATE_VM(mp_pending_exception) = MP_OBJ_NULL;
    pendsv_object = o;
    SCB->ICSR = SCB_ICSR_PENDSVSET_Msk;
```
This works fine unless mp_parse is interrupted, then we get the hardfaults (this happens on M7 MCUs F7 and H7).

I split the parse_compile_exec function in pyexec.c to do the parse_compile part with USB interrupts disabled, and the exec with USB interrupts enabled (so mp_parse does not get interrupted), and this seems to fix it.

dpgeorge, 2018-05-22:
> This works fine unless mp_parse is interrupted

Well, I don't really know why mp_parse() would behave differently to executing bytecode.  mp_parse() does raise exceptions just fine for SyntaxError's so the nlr_jump code is working in that context.

iabdalkader, 2018-05-22:
>> Well, I don't really know why mp_parse() would behave differently to executing bytecode. 

It doesn't, I meant that by splitting the parse_compile_exec I can run the parsing & compile bits with USB IRQ disabled so mp_parse doesn't get interrupted and pensv is not set during parsing, otherwise it hardfaults.

This is the scenario that triggers the hardfault: calling parse_compile_execute() and just before `mp_call_function_0(module_fun);` USB IRQ interrupts and sets pendsv, then it hardfaults.

iabdalkader, 2018-05-25:
Here's another (useful ?) trace:
```
Breakpoint 1, HardFault_C_Handler (regs=0x2001fc88) at stm32_it.c:144
144	    if (!pyb_hard_fault_debug) {
(gdb) backtrace 
#0  HardFault_C_Handler (regs=0x2001fc88) at stm32_it.c:144
#1  <signal handler called>
#2  nlr_jump (val=0x200117e0) at ../../py/nlrthumb.c:85
#3  0x0809399c in vstr_add_len (len=1, vstr=0x20011d04) at ../../py/vstr.c:125
#4  vstr_add_byte (vstr=0x20011d04, b=<optimized out>) at ../../py/vstr.c:142
#5  0x0809399c in vstr_add_len (len=1, vstr=0x20011d04) at ../../py/vstr.c:125
#6  vstr_add_byte (vstr=vstr@entry=0x20011d04, b=<optimized out>) at ../../py/vstr.c:142
#7  0x0807e206 in mp_lexer_to_next (lex=lex@entry=0x20011cc0) at ../../py/lexer.c:539
#8  0x08092e02 in mp_parse (lex=lex@entry=0x20011cc0, input_kind=input_kind@entry=MP_PARSE_FILE_INPUT) at ../../py/parse.c:948
#9  0x080a2ac4 in parse_compile_execute (source=<optimized out>, input_kind=input_kind@entry=MP_PARSE_FILE_INPUT, 
    exec_flags=exec_flags@entry=40) at ../../lib/utils/pyexec.c:93
#10 0x080a2e38 in pyexec_str (str=<optimized out>) at ../../lib/utils/pyexec.c:514
#11 0x0802ebca in main () at main.c:608
(gdb) print /x SCB->CFSR
$1 = 0x20000
```

RockySong, 2018-05-28:
It seems vstr_add_len can't call nlr_jump directly, could it be an openMV IDE USB debug command invoked in USB IRQ handler, re-entered vstr_add_len, and set pend PendSV IRQ which in turn did the long jump?

RockySong, 2018-06-06:
I found one case that can lead to INVSTATE hardfault:
If pendSV is serviced during an LDM/STM instruction, which is interruptible and uses XPSR bitfield (ICI/IT field) to save load/store progress, it can causes INVSTATE type hardfault when CPU finally returns to nlr_jump() as there is no corresponding LDM/STM instruction.
A workaround is manually modifying stacked XPSR at the beginning of pendsv handler (PendSV_Handler):
"mov r2,  #0x01000000 \n"
"str r2, [sp, #28] \n" // modify stacked XPSR to make sure possible LDM/STM progress is cleared

dpgeorge, 2024-10-14:
This has been fixed by ece950d9bed18554e945188c5a31908c8bfa9a7b, by removing preemptive NLR jump via PendSV.

closed 2024-10-14 by dpgeorge