micropython/micropython #3811: Issue with ports/stm32/usbdev/Release_Notes.html on clone

JoeSc, 2018-05-22:
When I perform a checkout of the micropython repository I get a modified file, 

```console
$ docker run -it ubuntu:18.04 /bin/bash
root@64f33d6e8c6f:/# apt-get update && apt-get install -y ca-certificates git
...
root@64f33d6e8c6f:/# git clone https://github.com/micropython/micropython.git micropython
Cloning into 'micropython'...
remote: Counting objects: 62704, done.
remote: Compressing objects: 100% (2/2), done.
remote: Total 62704 (delta 1), reused 1 (delta 1), pack-reused 62701
Receiving objects: 100% (62704/62704), 33.38 MiB | 4.63 MiB/s, done.
Resolving deltas: 100% (45917/45917), done.
root@64f33d6e8c6f:/# cd micropython/
root@64f33d6e8c6f:/micropython# git status
On branch master
Your branch is up to date with 'origin/master'.

Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git checkout -- <file>..." to discard changes in working directory)

	modified:   ports/stm32/usbdev/Release_Notes.html

no changes added to commit (use "git add" and/or "git commit -a")
root@64f33d6e8c6f:/micropython# 
```

As far as I can tell this is somehow related to CR/LF as when I try to do a git stash I get the following error
```console
root@64f33d6e8c6f:/micropython# git stash
warning: CRLF will be replaced by LF in ports/stm32/usbdev/Release_Notes.html.
The file will have its original line endings in your working directory.
Saved working directory and index state WIP on master: 18e63584 py/emit: Combine setup with/except/finally into one emit function.
root@64f33d6e8c6f:/micropython# 
```

Using git reset --hard to roll back one commit at a time, since I was unsure how to bisect this issue, I came upon commit 4b3c6290 as the culprit
``` console
root@64f33d6e8c6f:/micropython# git reset --hard HEAD~1; git status
HEAD is now at 4b3c6290 .gitattributes: Remove special text handling of stm32 usbdev files.
On branch master
Your branch is behind 'origin/master' by 78 commits, and can be fast-forwarded.
  (use "git pull" to update your local branch)

Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git checkout -- <file>..." to discard changes in working directory)

	modified:   ports/stm32/usbdev/Release_Notes.html

no changes added to commit (use "git add" and/or "git commit -a")
root@64f33d6e8c6f:/micropython# git reset --hard HEAD~1; git status
HEAD is now at abde0fa2 stm32/usbdev: Convert files to unix line endings.
On branch master
Your branch is behind 'origin/master' by 79 commits, and can be fast-forwarded.
  (use "git pull" to update your local branch)

nothing to commit, working tree clean
root@64f33d6e8c6f:/micropython# 
```

I believe this issue could be fixed by just doing a git commit of that file, but as I'm unsure of how best to fix this I wanted to first open this issue before I'd submit a pull request.

dpgeorge, 2018-05-23:
Thanks for the report.  I can confirm the problem.

IMO the offending Release_Notes.html file can just be deleted.  It doesn't really add anything, all the version and licensing info is repeated at the top of each relevant file in the ports/stm32/usbdev subdir.

dpgeorge, 2018-05-24:
Fixed by 4200018a05bcf7005c467245eb6c136f4fff9651

closed 2018-05-24 by dpgeorge