micropython/micropython #3813: ESP8285 can't use ap_if.config "can't set AP config"

mindforger, 2018-05-23:
i did the following

>ap_if = network.WLAN(network.AP_IF)
ap.active(False)
ap.config(essid="someID", password="somePassword")

but i always get "can't set AP config", no matter what parameter i try to use

MicroPython v1.9.4-8-ga9a3caad0 on 2018-05-11; ESP module with ESP8266 

flashed on an plain ES8285 Module

listdir is also doing funky stuff on this module, getting a crap ton of " '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00'," as answer + the folders and file that icreated for a test

mindforger, 2018-05-23:
how do i do that? i did not find a methode "format" yet in the docs

dpgeorge, 2018-05-28:
> stop, i CAN change the essid but i can not change the password

Did you try erasing completely the flash (esptool.py erase_flash, or "make erase") before putting MicroPython on it?

mindforger, 2018-05-28:
Yes i erased it, but on an 8285 it seems to break the default filesystem, i formated it and created a new empty Boot.py

fjand5, 2020-08-11:
you have to active befor config.

> 
> import network
> ap_if = network.WLAN(network.AP_IF)
> ap_if.active(False)
> ap_if.config(essid="someID", password="somePassword")
>

Medve01, 2022-03-05:
I have the same issue. It's a brand new board, flash erased before uploading firmware.
Board starts with a default AP set to "micropython-XXXX". I use the code from previous comment and I get the same error "OSError: can't set AP config".

I cannot change AP config...