micropython/micropython #3816: can not mount SDCard, it is hanging

mindforger, 2018-05-24:
if i try to mount an SDCard on my ESP8285 i can start up the SDCard interface but not mount the sd card itself

>import os, sdcard, machine
sd = sdcard.SDCard(machine.SPI(1), machine.Pin(16))
os.mount(sd, '/sd')

but it simply gets stuck at this last line

after ctrl+C it tells me it is hanging in the loop that is scanning for the first 0xfe

i always get back 0xff and the loop around line 177 is stuck forever

SDCard is connected to the appropriate pins, if i had any wiring issue, the first call would fail with an error (tested multiple times)

is there anything i can do to evaluate my SDcard is working without mounting it?

mindforger, 2018-05-25:
Okay, so i maybe found the issue

when the SD Card is too big (8GB) the mounting will hang up, maybe either add an sanity check on the card propertys before trying to seek infinite loops

dpgeorge, 2018-05-28:
> also there is a /lib in sys.path, i created the /lib folder with os.mkdir, is that okay?

Yes that's ok.  Also ok to not have /lib.

> is there anything i can do to evaluate my SDcard is working without mounting it?

Yes you can try reading (and writing if you don't care about its contents) blocks.  Eg:
```python
buf = bytearray(512)
sd.readblocks(0, buf)
print(buf)
```
and `buf` should look like the first sector, with 0x55aa at the end.

> PS the SDCard worked on an real ESP8266 before and it was FAT32 if i remember correct
> ...
> Okay, so i maybe found the issue
> when the SD Card is too big (8GB) the mounting will hang up

So, did you come to a conclusion as to the cause: is it a combination of ESP8285 and 8GB SD card that is problematic, or just 8GB on any device?

mindforger, 2018-05-28:
i tested 2 different 8gb cards, both won't work until i reduce the card size, i did not test the limit but somewhere between 2 and 4 gb it fails.. Maybe the same 3.2gb issue like win32? Maybe something loops infinite when doing the search around line 177

dpgeorge, 2018-05-29:
I tried a 4GB card with the sdcard.py driveron a PYBLITEv1.0 and it works OK.  The card is formatted to have 32768 bytes per sector and 120320 total sectors, giving a byte-size of 3942645760 bytes.

mindforger, 2018-05-30:
just had an idea, but i don't have te actual knowledge on how to do or even how compicated it would  be, but how about having a script to format an attached SD card? I mean there is already a builtin command to do so on the bedev, hwo can i format an attacjed SD properly using this route?

mindforger, 2018-06-29:
buggers, i don't have a setup to compile it myself, will it be released soon?

matikij, 2018-06-30:
@mindforger I don't think you need to compile it to test it out, perhaps you could try copyting the module from https://github.com/matikij/micropython/raw/fix-sdcard/drivers/sdcard/sdcard.py directly to '/ or '/lib' of board (possibly under another name) and then import and test this out.
As I don't really have that much cards to test this change it would be cool to know if it helps you as well.

mindforger, 2018-07-01:
oh wait i am stupid, i forgot this is already pyscript

dpgeorge, 2022-06-10:
Closing, it should be fixed by the above PR.

closed 2022-06-10 by dpgeorge