micropython/micropython #3818: (ESP32) How to list devices connected in AP_IF mode?

nickzoic, 2018-05-24:
This is copied across from https://github.com/micropython/micropython-esp32/issues/186 .

I had a look at this at the time and noted there was no RSSI information available to the AP.
That's since changed thanks to https://github.com/espressif/esp-idf/issues/1095

But I wanted to make this a new issue for MicroPython because we should try to get a standard status structure here. It would also be helpful, I think, to return the IP address information from the hostap along with MAC and RSSI and so on.  Perhaps, therefore, `if.status('stations')` should return a list of dicts with optional keys?

dpgeorge, 2018-06-05:
> It would also be helpful, I think, to return the IP address information from the hostap along with MAC and RSSI and so on.

Yes, IP address would be useful, but that's at a different network layer than the MAC/RSSI info.  Basically you want to query the status of the DHCP server.  Should that go in the `stations` status, or elsewhere?  Eventually we may want to be able to configure the DHCP server, like lease time and IP address range.

Furthermore, where would functionality go to deauth a connected station?  That's at the MAC layer and needs to be part of the NIC class (eg `nic.deauth(mac)`).  I know that's nothing really to do with getting a list of stations, but it's useful to think where that functionality would go to understand the scope of `if.status()` and `if.config()`.

> Perhaps, therefore, if.status('stations') should return a list of dicts with optional keys?

We have namedtuples (also their lightweight version attrtuple) which can do that quite well.  But extra features take RAM and code space so need to be considered.

nickzoic, 2018-06-05:
On Tue, Jun 5, 2018, at 15:28, Damien George wrote:
> layer than the MAC/RSSI info.  Basically you want to query the status
> of the DHCP server.  Should that go in the stations status, or
> elsewhere?  Eventually we may want to be able to configure the [...]
Ah, yeah, good point ... I wasn't really thinking about layers, just the
idea of being able to report on stations.  Maybe it would make sense to
have some way to query the DHCP server though.
> Furthermore, where would functionality go to deauth a connected
> station?  That's at the MAC layer and needs to be part of the NIC
> class (eg nic.deauth(mac)).  I know that's nothing really to do with [...]
yeah, we end up caught between "it returns a list of instances of class
'station', which has methods to ..." and wanting to keep the stack quite
light.  I think last time around this conversation we got kind of stuck
in the grey area between if.status('foo'), if.config('foo') and plain
old if.foo.
>> Perhaps, therefore, if.status('stations') should return a list of
>> dicts with optional keys?> We have namedtuples (also their lightweight version attrtuple) which
> can do that quite well.  But extra features take RAM and code space so
> need to be considered.
Good point.  How do you feel about those for returning mac and rssi at
least (rather than the current 2-tuple)

dpgeorge, 2018-06-05:
> How do you feel about those for returning mac and rssi at least (rather than the current 2-tuple)

Yes, I think that was already agreed upon and the docs do state it'll return tuples of the form (MAC, RSSI).