micropython/micropython #3819: Latest commit cannot compile

rolandvs, 2018-05-25:
Just updated my fork and after make it ends up in:

```
make: *** No rule to make target `lib/lwip/src/core/inet_chksum.c', needed by `build-PYBV10/genhdr/qstr.i.last'.  Stop.
```

dpgeorge, 2018-05-26:
You probably need to run a `git submodule update --init`

rolandvs, 2018-05-26:
That doesn't seem to work :-(

`py.mk`  has `core/ipv4/inet_chksum.c` and directory contains the file...

peterhinch, 2018-05-27:
I can't replicate this. The build scripts I use may be found [here](https://github.com/peterhinch/micropython-samples.git) `buildnew` followed by `buildpyb`.

dpgeorge, 2018-06-05:
@rolandvs did you solve this problem?

robert-hh, 2018-06-05:
I had this problem too, but "git submodule upate --init" solved that. I regularly forget that command.

rolandvs, 2018-06-05:
I tried the above, but no succes. However due to time constraints I skipped this and used a fresh copy. That works fine. So I probably dit something wrong while updating from git. If it surfaces again I will make an effort to track it down.

dpgeorge, 2018-06-06:
> So I probably dit something wrong while updating from git

Ok, so I'll close the issue then.  Feel free to reopen if neede.

closed 2018-06-06 by dpgeorge