micropython/micropython #3827: H743Zi with Current ARM compiler fails

K2AAE, 2018-05-30:
LINK build-NUCLEO_H743ZI/firmware.elf
arm-none-eabi-ld: error: build-NUCLEO_H743ZI/firmware.elf uses VFP register arguments, /usr/lib/gcc/arm-none-eabi/4.9.3/libgcc.a(bpabi.o) does not
arm-none-eabi-ld: failed to merge target specific data of file /usr/lib/gcc/arm-none-eabi/4.9.3/libgcc.a(bpabi.o)
arm-none-eabi-ld: error: build-NUCLEO_H743ZI/firmware.elf uses VFP register arguments, /usr/lib/gcc/arm-none-eabi/4.9.3/libgcc.a(_divdi3.o) does not
arm-none-eabi-ld: failed to merge target specific data of file /usr/lib/gcc/arm-none-eabi/4.9.3/libgcc.a(_divdi3.o)
arm-none-eabi-ld: error: build-NUCLEO_H743ZI/firmware.elf uses VFP register arguments, /usr/lib/gcc/arm-none-eabi/4.9.3/libgcc.a(_udivdi3.o) does not
arm-none-eabi-ld: failed to merge target specific data of file /usr/lib/gcc/arm-none-eabi/4.9.3/libgcc.a(_udivdi3.o)
Makefile:509: recipe for target 'build-NUCLEO_H743ZI/firmware.elf' failed
make: *** [build-NUCLEO_H743ZI/firmware.elf] Error 1

How I got the compiler:
sudo apt-get install gcc-arm-none-eabi binutils-arm-none-eabi gdb-arm-none-eabi openocd
5/30/2018

dpgeorge, 2018-05-31:
It looks like the linker is not finding/using the correct library for the MCU architecture.  The relevant line in ports/stm32/Makefile that finds libgcc.a is:
```
LIBS = $(shell $(CC) $(CFLAGS) -print-libgcc-file-name)
```
If you run `make BOARD=NUCLEO_H743ZI V=1` it should print near the end the path to libgcc.a that it's using.  On my system (Arch Linux arm-none-eabi-gcc v8.1.0) it's using:
```
$ arm-none-eabi-gcc -mthumb -mfpu=fpv5-d16 -mfloat-abi=hard -mcpu=cortex-m7 -print-libgcc-file-name
/usr/lib/gcc/arm-none-eabi/8.1.0/thumb/v7e-m+dp/hard/libgcc.a
```

dpgeorge, 2018-08-13:
I assume this was resolved.  Feel free to reopen with additional details if not.

closed 2018-08-13 by dpgeorge