micropython/micropython #3828: urequests ENOMEM after soft-reboot with workaround

cosmicc, 2018-05-31:
I was making a post with urequests, and I was noticing after a request was sent, then after a soft reboot I would get an error 12 ENOMEM error on the next post from urequests.  I suspected I wasn't closing the connection or something (which still may be the case, i'm new).  I also suspected some kind of memory not clearing properly.

Then I started playing with the garbage collector.  I noticed 2 things: after the ENOMEM error, if i ran a gc.collect() the device would reboot.  But, If I ran a gc.collect() after a successful urequests post, i could run another urequests post after a soft reboot!  Apparently, it collected the garbage away.

Could anyone enlighten on this?  Is this a bug? am i doing it wrong? or is gc.collect() something that should be run like that?

I'm running this on a feather huzzah32 with the esp32 build, if that matters.

dpgeorge, 2018-05-31:
Thanks for the report.  I can confirm the bug with crash after ENOMEM.  To reproduce do:
```python
>>> import ssl
>>> ssl.wrap_socket(None)
>>> ssl.wrap_socket(None) # should give ENOMEM
>>> import gc
>>> gc.collect() # may need to do this a few times to get a crash
```
This is a bug that should be fixed.

> I suspected I wasn't closing the connection or something

SSL sockets require a lot of memory and you need to close them when finished with them before opening another one.

dpgeorge, 2018-05-31:
The crash doing a gc.collect() should now be fixed by 98b9f0fc9d0fe14c5f13faf2e9b902422919594c

cosmicc, 2018-06-01:
I just built and flashed the latest (ea22406).  I can confirm the gc.collect() after the ENOMEM does not crash now.  But, the gc_sweep_all you added doesn't seem to clear it still after the soft reboot.  I did not call any functions, from what it looked like you are calling that gc_sweep_all on the soft reboot.  I still have my workaround doing a gc.collect() but I thought I would let you know, if that wasn't as intended.

dpgeorge, 2018-06-01:
> But, the gc_sweep_all you added doesn't seem to clear it still after the soft reboot.

This change was not pushed to master, it's just in a PR at the moment for review.

cosmicc, 2018-06-01:
I just pulled and built the PR, it worked. thanks D

dpgeorge, 2018-06-01:
> I just pulled and built the PR, it worked

Great, thanks for confirming!

dpgeorge, 2018-06-12:
Sweep-all was added in b2fa1b50edaea43007d692d24391c6f3aa4d15ca to help reclaim memory on a soft reset.

closed 2018-06-12 by dpgeorge