micropython/micropython #3833: enhancement - add API to save/restore state of virtual machine

adritium, 2018-06-01:
[Original question at the forum](https://forum.micropython.org/viewtopic.php?f=3&t=4855)

Saving/restoring the state requires saving/restoring two things: heap and `mp_state_ctx_t mp_state_ctx;`.
Unfortunately it's not as simple as writing/reading those two items to/from persistent storage.

When you restore, the new heap that you create might not be in same memory location as the old heap which means that ``mp_state_ctx_t mp_state_ctx;` must be modified because it contains pointers to location of heap.

Furthermore, the heap could even be of a different size and even a smaller heap is ok in theory (not sure about this).

dhylands, 2018-06-01:
Restoring the heap to a different location is basically the same problem as supporting a precise GC. So basically I think it would only work if being restored to same location and size.

The other big issue is the state of the hardware and all of the peripherals.

And you'd probably need to save the stack as well, although that may not be required if saving the state is only allowed from certain locations.

The heap may also contain pointers back into flash, so you wouldn't be able to save the state, update some firmware and restore the state.

dpgeorge, 2018-06-02:
This is somewhat related to #3619 -- if you could pause execution and return to the top level of the C code, then you'd have some chance of saving the VM state.

dpgeorge, 2022-06-23:
Closing due to inactivity.

closed 2022-06-23 by dpgeorge

rversteegen, 2022-06-24:
May I ask, is there still a desire or intention to implement this feature eventually?
(I'm most interested in #3619 though)