micropython/micropython #3834: pyboard - deepsleep wake pins

diginfo, 2018-06-02:
I have just switched to pyboard for a project which has been on the esp32 and I need to be able to wake from deep sleep by either time or a pin edge.

https://github.com/micropython/micropython/blob/master/docs/library/machine.Pin.rst
https://docs.micropython.org/en/latest/pyboard/library/machine.Pin.html

These pages indicate that the syntax for enabling a wake-up pin is:

```
import machine as mc
door = mc.Pin('X1',mc.Pin.IN)
door.irq(handler=None, trigger=(mc.Pin.IRQ_FALLING | mc.Pin.IRQ_RISING),wake=mc.DEEPSLEEP)
```

However this results in an error:

```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'module' object has no attribute 'DEEPSLEEP'
```

What's the solution ?

diginfo, 2018-06-02:
Thanks Damien, that works, pull down and active high is not a problem at all, might be worthwhile to add that to the docs at some stage.

I can link other pins to this one using a forward-diode and then check on wakeup which pin caused the reset, as in this application I need 2 wakeup pins.

But is there any non-volatile memory that survives deepsleep wakeup like the rtc.memory() on esp devices as I need to store some json string data ?

dpgeorge, 2018-06-02:
> in this application I need 2 wakeup pins.

I think PC13 (X18) can also wake up the MCU from deepsleep but I don't have any code samples for that right now.

> is there any non-volatile memory that survives deepsleep wakeup

Yes there is 4k of backup SRAM, at address 0x40024000.  To enable it use:
```python
import stm, uctypes
stm.mem32[stm.RCC+stm.RCC_AHB1ENR] |= 1<<18 # turn on BKPSRAMEN
bkp_sram = uctypes.bytearray_at(0x40024000, 4096)
```
Then use `bkp_sram` to save your data.

diginfo, 2018-06-03:
Hi Damien;

I have converted all of my code from esp32 to pyboard and all is now working (including non-volatile ram) except for this wake on interrupt.

Here's the file:

```
import machine as mc, stm, time; 
stm.mem32[stm.PWR + stm.PWR_CSR] |= 1 << 8; 
time.sleep(1); 
mc.deepsleep() 
```

**Here's the logs:**

```
>>> import wake
MicroPython v1.9.4 on 2018-05-11; PYBv1.1 with STM32F405RG
Type "help()" for more information.
>>> import wake
MicroPython v1.9.4 on 2018-05-11; PYBv1.1 with STM32F405RG
Type "help()" for more information.
>>> import wake
MicroPython v1.9.4 on 2018-05-11; PYBv1.1 with STM32F405RG
Type "help()" for more information.
>>> import wake
MicroPython v1.9.4 on 2018-05-11; PYBv1.1 with STM32F405RG
Type "help()" for more information.
>>> import wake
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "wake.py", line 1, in <module>
NameError: name '���������������������������������������������������������������������������������������������������������������' is not defined
>>> 
```

After calling & waking the device a few times, the file system gets corrupted, in this case the wake.py file is corrupted, main.py is corrupted and what were folders are zero-byte files.

Any idea what is happening ??

diginfo, 2018-06-03:
OK, I think I have tracked this down to usb storage mode causing file system corruption.

when I disable storeage mode:

`pyb.usb_mode('VCP')`

The problem does not occur.

dpgeorge, 2018-06-04:
> The backup RAM referenced above (being SRAM) requires a backup button cell.

It should still work (be retained) without a backup battery (Vbat) if you enable the BRE bit:
```python
stm.mem32[stm.PWR + stm.PWR_CSR] |= 1 << 9 # BRE backup regulator enable
```

> OK, I think I have tracked this down to usb storage mode causing file system corruption.

Yes, that's most likely the problem.  When it goes to deepsleep the USB will be shut off.  So best not to use USB at all (which is anyway power hungry).

diginfo, 2018-06-04:
Created this jsor r/w for use on esp32 / pyboard based on info gathered from the forum:

**jram.py**
```
import json
from sys import platform

if platform == 'pyboard':
  
  import stm, uctypes
  
  class JRAM(object):
    BKPSRAM = 0x40024000
    def __init__(self):
      stm.mem32[stm.RCC + stm.RCC_APB1ENR] |= 0x10000000    # PWREN bit
      stm.mem32[stm.PWR + stm.PWR_CR] |= 0x100              # Set the DBP bit in the PWR power control register
      stm.mem32[stm.RCC +stm.RCC_AHB1ENR]|= 0x40000         # enable BKPSRAMEN
      stm.mem32[stm.PWR + stm.PWR_CSR] |= 0x200             # BRE backup register enable bit
    
    def __getitem__(self, idx):
      assert idx >= 0 and idx <= 0x3ff, "Index must be between 0 and 1023"
      return stm.mem32[self.BKPSRAM + idx * 4]
    
    def __setitem__(self, idx, val):
      assert idx >= 0 and idx <= 0x3ff, "Index must be between 0 and 1023"
      stm.mem32[self.BKPSRAM + idx * 4] = val
    
    def get_bytearray(self):
      return uctypes.bytearray_at(self.BKPSRAM, 4096)
  
    def clr(self):
      self.put({})
      return self.get()  
      
    def get(self):
      ba = self.get_bytearray()
      try:
        return json.loads(bytes(ba[4:4+self[0]]).decode("utf-8"))
      except:
        return self.clr()
        
    def put(self,a):
      ba = self.get_bytearray()
      z = json.dumps(a).encode('utf8')
      self[0] = len(z)
      ba[4: 4+len(z)] = z

elif platform.startswith('esp32'):
  from machine import RTC
  rtc = RTC()
  
  class JRAM():

    ## Non Volatile JSON
    def get(self):
      try:return json.loads(rtc.memory().decode('ascii'))
      except: return {}
    
    def put(self,dic):
      rtc.memory(json.dumps(dic))
  
    def clr(self):
      put({})
      return get()
    
## usage
# from jram import JRAM as nvr
# nvr().clr()
# nvr().put({'elephants': 9, 'rats': 77, 'dogs': 99, 'zoo': 100})
# nvr().get()
```

Lisa999, 2018-08-09:
Bugs on ESP32:
`def clr(self):
      put({})
      return get()`
Should be self.put()
import json should be ujson.
On ESP32 it doesn't work because the esp clears the RTC mem during startup.

pacmac, 2019-04-13:
I adapted this for the PYBOARD-D ( hasattr(stm,'PWR_CR1') ) however it appears to break the RTC().wakeup() function used elsewhere, basically it never wakes up.

Any idea why ??

```
  import stm, uctypes
  
  class JRAM(object):
    BKPSRAM = 0x40024000
    def __init__(self):
      if hasattr(stm,'PWR_CR1'):    
        stm.mem32[stm.RCC + stm.RCC_APB1ENR] |= 0x10000000    # PWREN bit
        stm.mem32[stm.PWR + stm.PWR_CR1] |= 0x100             # Set the DBP bit in the PWR power control register
        stm.mem32[stm.RCC +stm.RCC_AHB1ENR]|= 0x40000         # enable BKPSRAMEN
        stm.mem32[stm.PWR + stm.PWR_CSR1] |= 0x200            # BRE backup register enable bit
      else:
        stm.mem32[stm.RCC + stm.RCC_APB1ENR] |= 0x10000000    # PWREN bit
        stm.mem32[stm.PWR + stm.PWR_CR] |= 0x100              # Set the DBP bit in the PWR power control register
        stm.mem32[stm.RCC +stm.RCC_AHB1ENR]|= 0x40000         # enable BKPSRAMEN
        stm.mem32[stm.PWR + stm.PWR_CSR] |= 0x200             # BRE backup register enable bit    
```

dpgeorge, 2019-04-18:
> I adapted this for the PYBOARD-D ( hasattr(stm,'PWR_CR1') ) however it appears to break the RTC().wakeup() / deepsleep() function used elsewhere, basically it never wakes up.
> 
> Any idea why ??

This turned out to be an errata of the F7 MCU.  Apparently one must set the `PWR_CSR1_EIWUP` bit.  Adding `stm.mem32[stm.PWR + stm.PWR_CSR1] |= 0x100` after enabling BRE fixes it.

I've applied this fix in 8402c26cfa98b4689f5ac4673952a654cfe5b678