micropython/micropython #3840: ESP32:  no default gc threshold

MikeTeachman, 2018-06-05:
Latest stable firmware (esp32-20180511-v1.9.4.bin) does not appear to have a  garbage collection threshold.  running gc.threshold() in the REPL returns -1.    gc.isenabled() returns True.  Running in a loop will take the free heap space to zero, with no automatic gc.  

Is there a reason for this default configuration?   The esp8266 has an out-of-the-box gc threshold of 8992.

dpgeorge, 2018-06-05:
> Running in a loop will take the free heap space to zero, with no automatic gc.

Once free space drops to zero it will do a collection automatically.  If you want to collect earlier than this (to help prevent heap fragmentation) you can se the GC threshold in your boot.py/main.py/etc.

MikeTeachman, 2018-06-05:
Thanks.   I'm still curious to know the design reasons for having the ESP32 GC threshold settings different than the ESP8266.  The ESP8266 has a GC threshold = 8992 in the latest stable release.

tve, 2020-05-21:
I bumped into this issue by chance. I have an app that runs with ~30KB of free space on the esp32. It communicates a lot via MQTT. Without gc threshold it sooner or later dies due to heap fragmentation. I set the threshold to 4096 and that cured the problem. I think the current no-threshold setting is a very poor choice, it forces everyone to hit that wall and look for the solution without providing any real benefit.

Along the way I tried a couple of things:
- I found out about CP's extensive changes to GC and got jealous
- I added a gc.collect() call after every import (in builtinimport.c) and found that I got a lot of the benefits that CP's changes got
- I set the gc threshold in boot.py to 1024, 4096, or 8192 and found that 4096 accomplishes all I need and 1024 isn't much better but uses more GC time, 8192 is noticeably worse

The only reason I haven't created a PR is PSRAM. With a 2GB heap GC takes a long time. Options considered:
- set the threshold to 4% of heap instead of 4KB
- start with a smaller heap when using PSRAM and expand it when needed (could allocate all initially from esp-idf but only allow MP to see a portion), this way GC time stays more under control
- only set threshold when not using PSRAM

I think the first or last option, even though neither is great, would be an overall enhancement. It would be better if users that don't like it have to call gc.threshold than seeing new users that have no clue about how GC works run into the fragmentation wall. IMHO...

(Only tangentially related, it seems time for MP to overhaul its allocator and GC, but that's obviously a bigger story...)

jonnor, 2024-09-29:
Thanks for the testing and suggestions @tve 

I believe that MicroPython on ESP32 now does default to a smaller GC heap on PSRAM. So maybe that makes it more feasible to set a sane default here?