micropython/micropython #3841: Pyboard - CC3200 WIFI

diginfo, 2018-06-05:
Hi;

I am looking at enabling WIFI on the pyboard, specifically with CC3200 Module, but before I go ahead and order a bunch of them, can you confirm the required pins ?

I see that clk, cs, miso and mosi are GPIO14 - 17, but what about VBEN and IRQ ??

[image]

Also, is the 3200 compatible with the 3000 ?

dpgeorge, 2018-06-05:
> is the 3200 compatible with the 3000

No, it definitely isn't!  And I wouldn't recommend using the CC3000.  There is some support for the CC3100 module as external WiFi support but it's not in mainline (see eg https://github.com/emfcamp/micropython but that fork is quite old now).

diginfo, 2018-06-05:
Damien, is there any current / proven support for a wlan connection on the pyboard ?

diginfo, 2018-06-05:
I have succeeded in connecting a (firmware restored) 8266 using the code from here: http://www.tinyosshop.com/download/PYWIFI.zip - but how could this be integrated to work with nthe networking / wlan modules ?

untitledlt, 2019-11-03:
is there any progress on this?

dpgeorge, 2022-08-26:
There is no plan to support CC3200 as an external WiFi chip.  Support for CC3000 was removed in 1855df6361ad4ade62aa46ad08f3202a6a5b1bd1

There are many other options for WiFi, eg PYBD boards, Pico W, or using an esp32 as a coprocessor (Ublox Nina).

closed 2022-08-26 by dpgeorge