micropython/micropython #3842: Cortex-M cores: pendSV handler need to modify the auto stacked XPSR value to clear LDM/STM progress

RockySong, 2018-06-05:
If pendSV is serviced during an LDM/STM instruction, which is interruptible and uses XPSR bitfields to save load/store progress, it may causes INVSTATE type hardfault when CPU finally returns to nlr_jump().
This is because the Cortex-M core is expecting to resume LDM/STM when it sees the valid progress information in unstacked XPSR register 
However, pendSV returns to nlr_jump(). So there is no LDM/STM instruction and the Cortex-M CPU generates hardfault with INVSTATE.
My workaround, insert below instructions in the beginning of PendSV_Handler:
		"mov r2,	#0x01000000 \n"
		"str r2, [sp, #28]      \n"    // modify stacked XPSR to make sure possible LDM/STM progress is cleared

So when nlr_jump() is entered, it always has XPSR unstacked to a "clean" state.

Note: This may have something to do with the INVSTATE hardfault found in openMV project.

iabdalkader, 2018-06-05:
Is this going to break multithreading ? BTW another solution is to disable interruption of multicycle instructions:

> DISMCYCINT : Disables interruption of multi-cycle instructions. This increases the interrupt latency of the processor because load/store and multiply/divide operations complete before interrupt stacking occurs

I'm almost sure I tried this before but didn't work (maybe I didn't upload the right fw image when testing)

RockySong, 2018-06-06:
I did captured real case that LDM/STM is interrupted and ICI field saved its progress. I think DISMCYCINT should also work as it makes LDM/STM can't be interrupted.

iabdalkader, 2018-06-09:
@dpgeorge Any comment on this issue ?

xiabizha, 2019-08-20:
would any plan of micropython to support resolve this issue?

dpgeorge, 2024-10-14:
This has been fixed by ece950d9bed18554e945188c5a31908c8bfa9a7b, by removing preemptive NLR jump via PendSV.

closed 2024-10-14 by dpgeorge