micropython/micropython #3843: Free resource before raise syntax error so we can access python file after syntax error.

KrisJe, 2018-06-05:
https://github.com/micropython/micropython/blob/aace60a75e5afedadfc30d33e91ef7c6fa312405/py/parse.c#L1156

        mp_obj_exception_add_traceback(exc, lex->source_name, lex->tok_line, MP_QSTR_NULL);
        mp_lexer_free(lex); //<-- add this
        nlr_raise(exc);

dpgeorge, 2018-06-09:
Thanks for the suggestion.  What was it exactly that prompted this change?

tomlogic, 2018-06-17:
@dpgeorge, I was about to open an issue for this problem, so I can provide additional information on why I think it's a problem.

`mp_lexer_free()` calls `lex->reader.close(lex->reader.data);`.  When parsing a file on disk (e.g., `foo.py` as a result of `import foo`), we end up keeping a handle open to that file if throwing an exception in `mp_parse()` or `fold_constants()`.  I'm not sure if other functions called from `mp_parse()` can throw an exception or not.  Perhaps it's safest if `mp_parse()` has a wrapper that calls `mp_lexer_free()` and re-throws the exception?  I'd like some feedback on that idea before actually implementing it, though.  It seems "right" since `mp_parse()` already takes responsibility for freeing the lexer (and closing the reader) on success.

I have a similar fix I plan to submit for `mp_raw_code_load()`, the equivalent `import` code path for loading `.mpy` files.

dpgeorge, 2023-09-14:
Fixed by 5e122b11eaee64bbece2ace08fc3f994d5f83f2e

closed 2023-09-14 by dpgeorge