micropython/micropython #3844: bug - compile error - MICROPY_ENABLE_GC==0 && MICROPY_ENABLE_FINALISER==1

adritium, 2018-06-05:
https://github.com/micropython/micropython/blob/aace60a75e5afedadfc30d33e91ef7c6fa312405/py/malloc.c#L102-L107

If `MICROPY_ENABLE_GC==0`, should this code be allowed to be compiled even if `MICROPY_ENABLE_FINALISER==1`?

It causes a compile error.

stinos, 2018-06-05:
What error exactly? The int to void* conversion?

adritium, 2018-06-05:
`malloc.c(101,5): error : implicit declaration of function 'malloc_with_finaliser' [-Werror=implicit-function-declaration]`

adritium, 2018-06-19:
I was implying - not strongly enough - that usage of finaliser macro should be tied to gc macro.

dpgeorge, 2018-06-20:
@adritium do you have a suggestion what to do here, how to resolve the issue?

adritium, 2018-07-02:
@dpgeorge 

> It's not really supported to have finilisation enabled with the GC disabled

Let's add `MICROPY_ENABLE_GC == 1` condition such that

```C
#if MICROPY_ENABLE_FINALISER &&  MICROPY_ENABLE_GC
 void *m_malloc_with_finaliser(size_t num_bytes) { 
     void *ptr = malloc_with_finaliser(num_bytes); 
     if (ptr == NULL && num_bytes != 0) { 
         m_malloc_fail(num_bytes); 
     }
```

dpgeorge, 2018-07-03:
> Let's add MICROPY_ENABLE_GC == 1 condition such that

But that would just push the error message to a link error, like: `vfs_fat_file.c:(.text+0x6ba): undefined reference to m_malloc_with_finaliser`

Since it's not supported to do this I think it'd be better to emit macro error (using `#error`) if the GC is disabled with finalisers enabled.

adritium, 2018-07-03:
What’s different between this issue and #3914 ?
In both there is a compile-time issue when one mpconfig.h macro is enabled and another one isn’t. 

Unless this one is different, I thought it’d make sense to have a consistent scheme: if a piece of code requires one or more mpconfig.h macros to properly compile/link then surround said code with ALL of them. 

But you know the code better; if this issue is an exception, please say why.

dpgeorge, 2018-07-09:
Fixed by fcf621b066142cc221e5357f54e8156b1cb8c7fd

closed 2018-07-09 by dpgeorge