micropython/micropython #3845: Cannot build for OpenWRT: skipping incompatible build/libaxtls.a

dimka-rs, 2018-06-05:
I'm trying to build v1.9.4 for OpenWRT 15.05.
Most important part of package Makefile looks like this:
```
MAKE_FLAGS += \
       -C $(PKG_BUILD_DIR)/ports/unix \
       MICROPY_USE_READLINE=1

define Package/micropython/install
    $(INSTALL_DIR) $(1)/usr/bin
    $(INSTALL_BIN) $(PKG_BUILD_DIR)/ports/unix/micropython $(1)/usr/bin/micropython
endef

define Build/Prepare
    $(Build/Prepare/Default)
    echo "===== PREPARE ====="
    (cd $(PKG_BUILD_DIR) && git submodule update --init)
    echo "===== MPY-CROSS ====="
    (cd $(PKG_BUILD_DIR) && $(MAKE) -C mpy-cross)
    echo "===== DEPLIBS ====="
    (cd $(PKG_BUILD_DIR)/ports/unix && $(MAKE) deplibs)
    #echo "===== AXTLS ====="
    #(cd $(PKG_BUILD_DIR)/ports/unix && $(MAKE) axtls)
endef

$(eval $(call BuildPackage,micropython))
```

Build fails with
```
mipsel-openwrt-linux-uclibc/bin/ld: skipping incompatible build/libaxtls.a when searching for -laxtls
mipsel-openwrt-linux-uclibc/bin/ld: cannot find -laxtls
```
Why could build/libaxtls.a possibly be incompatible?
Build succeeds with MICROPY_PY_USSL = 0

Full linker invocation:
```
LINK micropython
mipsel-openwrt-linux-uclibc-gcc -o micropython build/py/mpstate.o build/py/nlr.o build/py/nlrx86.o build/py/nlrx64.o build/py/nlrthumb.o build/py/nlrxtensa.o build/py/nlrsetjmp.o build/py/malloc.o build/py/gc.o build/py/pystack.o build/py/qstr.o build/py/vstr.o build/py/mpprint.o build/py/unicode.o build/py/mpz.o build/py/reader.o build/py/lexer.o build/py/parse.o build/py/scope.o build/py/compile.o build/py/emitcommon.o build/py/emitbc.o build/py/asmbase.o build/py/asmx64.o build/py/emitnx64.o build/py/asmx86.o build/py/emitnx86.o build/py/asmthumb.o build/py/emitnthumb.o build/py/emitinlinethumb.o build/py/asmarm.o build/py/emitnarm.o build/py/asmxtensa.o build/py/emitnxtensa.o build/py/emitinlinextensa.o build/py/formatfloat.o build/py/parsenumbase.o build/py/parsenum.o build/py/emitglue.o build/py/persistentcode.o build/py/runtime.o build/py/runtime_utils.o build/py/scheduler.o build/py/nativeglue.o build/py/stackctrl.o build/py/argcheck.o build/py/warning.o build/py/map.o build/py/obj.o build/py/objarray.o build/py/objattrtuple.o build/py/objbool.o build/py/objboundmeth.o build/py/objcell.o build/py/objclosure.o build/py/objcomplex.o build/py/objdeque.o build/py/objdict.o build/py/objenumerate.o build/py/objexcept.o build/py/objfilter.o build/py/objfloat.o build/py/objfun.o build/py/objgenerator.o build/py/objgetitemiter.o build/py/objint.o build/py/objint_longlong.o build/py/objint_mpz.o build/py/objlist.o build/py/objmap.o build/py/objmodule.o build/py/objobject.o build/py/objpolyiter.o build/py/objproperty.o build/py/objnone.o build/py/objnamedtuple.o build/py/objrange.o build/py/objreversed.o build/py/objset.o build/py/objsingleton.o build/py/objslice.o build/py/objstr.o build/py/objstrunicode.o build/py/objstringio.o build/py/objtuple.o build/py/objtype.o build/py/objzip.o build/py/opmethods.o build/py/sequence.o build/py/stream.o build/py/binary.o build/py/builtinimport.o build/py/builtinevex.o build/py/builtinhelp.o build/py/modarray.o build/py/modbuiltins.o build/py/modcollections.o build/py/modgc.o build/py/modio.o build/py/modmath.o build/py/modcmath.o build/py/modmicropython.o build/py/modstruct.o build/py/modsys.o build/py/moduerrno.o build/py/modthread.o build/py/vm.o build/py/bc.o build/py/showbc.o build/py/repl.o build/py/smallint.o build/py/frozenmod.o build/extmod/moductypes.o build/extmod/modujson.o build/extmod/modure.o build/extmod/moduzlib.o build/extmod/moduheapq.o build/extmod/modutimeq.o build/extmod/moduhashlib.o build/extmod/modubinascii.o build/extmod/virtpin.o build/extmod/machine_mem.o build/extmod/machine_pinbase.o build/extmod/machine_signal.o build/extmod/machine_pulse.o build/extmod/machine_i2c.o build/extmod/machine_spi.o build/extmod/modussl_axtls.o build/extmod/modussl_mbedtls.o build/extmod/modurandom.o build/extmod/moduselect.o build/extmod/modwebsocket.o build/extmod/modwebrepl.o build/extmod/modframebuf.o build/extmod/vfs.o build/extmod/vfs_reader.o build/extmod/vfs_fat.o build/extmod/vfs_fat_diskio.o build/extmod/vfs_fat_file.o build/extmod/utime_mphal.o build/extmod/uos_dupterm.o build/lib/embed/abort_.o build/lib/utils/printf.o build/build/frozen.o build/build/frozen_mpy.o build/main.o build/gccollect.o build/unix_mphal.o build/mpthreadport.o build/input.o build/file.o build/modmachine.o build/modos.o build/moduos_vfs.o build/modtime.o build/moduselect.o build/alloc.o build/coverage.o build/fatfs_port.o build/extmod/modbtree.o build/lib/berkeley-db-1.xx/btree/bt_close.o build/lib/berkeley-db-1.xx/btree/bt_conv.o build/lib/berkeley-db-1.xx/btree/bt_debug.o build/lib/berkeley-db-1.xx/btree/bt_delete.o build/lib/berkeley-db-1.xx/btree/bt_get.o build/lib/berkeley-db-1.xx/btree/bt_open.o build/lib/berkeley-db-1.xx/btree/bt_overflow.o build/lib/berkeley-db-1.xx/btree/bt_page.o build/lib/berkeley-db-1.xx/btree/bt_put.o build/lib/berkeley-db-1.xx/btree/bt_search.o build/lib/berkeley-db-1.xx/btree/bt_seq.o build/lib/berkeley-db-1.xx/btree/bt_split.o build/lib/berkeley-db-1.xx/btree/bt_utils.o build/lib/berkeley-db-1.xx/mpool/mpool.o build/modtermios.o build/modusocket.o build/modffi.o build/lib/mp-readline/readline.o build/lib/timeutils/timeutils.o build/lib/oofatfs/ff.o build/lib/oofatfs/option/unicode.o  -Lbuild -laxtls -lpthread -L/home/dimka/repos/openwrt-15_05/staging_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2_R2/usr/lib -lffi -ldl -Wl,-Map=micropython.map,--cref -Wl,--gc-sections -lm 
/home/dimka/repos/openwrt-15_05/staging_dir/toolchain-mipsel_24kec+dsp_gcc-4.6-linaro_uClibc-0.9.33.2/lib/gcc/mipsel-openwrt-linux-uclibc/4.6.4/../../../../mipsel-openwrt-linux-uclibc/bin/ld: skipping incompatible build/libaxtls.a when searching for -laxtls
/home/dimka/repos/openwrt-15_05/staging_dir/toolchain-mipsel_24kec+dsp_gcc-4.6-linaro_uClibc-0.9.33.2/lib/gcc/mipsel-openwrt-linux-uclibc/4.6.4/../../../../mipsel-openwrt-linux-uclibc/bin/ld: cannot find -laxtls
collect2: ld returned 1 exit status
../../py/mkrules.mk:128: recipe for target 'micropython' failed
make[3]: *** [micropython] Error 1
```

dhylands, 2018-06-05:
It's probably built for a different architecture? You can find out what architecture it is using the 'file' command. The libtls.a file should show as an archive. You can extract one of the .o files:
```
ar x build/libaxtls.a aes.o
file aes.o
```
On my linux build machine, it shows as:
```
2017 >file aes.o
aes.o: ELF 64-bit LSB relocatable, x86-64, version 1 (SYSV), not stripped
```
You want it to show up as the architecture that your OpenWRT build will be running on (it shows as x86-64 above).

dhylands, 2018-06-05:
It looks like libaxtls doesn't have a special build directory per architecture and there is no clean target as part of the micropython build. So you may need to do:
```
cd ../../lib/axtls
make clean
```
before building, to make sure any .o files from a different architecture get removed,

dimka-rs, 2018-06-06:
@dhylands you are right, I'm building for MIPS, but library is for x86-64
```
aes.o: ELF 64-bit LSB relocatable, x86-64, version 1 (SYSV), not stripped
```
Something went wrong with cross compilation.
Thank you!

dpgeorge, 2018-06-09:
I assume this was resolved by rebuilding all source for the right architecture.

closed 2018-06-09 by dpgeorge