micropython/micropython #3852: Found bug related to the pybcdc.inf file.

kwagyeman, 2018-06-09:
Note that on the latest releases of windows 10 the pybcdc.inf file is no longer needed... and if you install it will turn the pyboard's MSC drive into a uart on windows.

Ibrahim fixed this issue for the OpenMV Cam. However, I don't know which file he fixed for this issue. The problem is related to some of the descriptor settings in the USB code. If you're interested in fixing this let me know and I'll get Ibrahim to find the commit me made to fix this issue.

A user can fix this issue by uninstalling and deleting the device files for the USB serial ports under the device manager on any windows versions affected by this issue and things will work.

kwagyeman, 2018-06-09:
Here's the fix: https://github.com/openmv/micropython/commit/daf8a0e6c1039eeb7e3db3348b781d9a40500716#diff-f3d986c7ee10da544654e4ebd8716d83

kwagyeman, 2018-06-09:
Maybe this is it? https://github.com/openmv/micropython/commit/daf8a0e6c1039eeb7e3db3348b781d9a40500716#diff-f3d986c7ee10da544654e4ebd8716d83

@iabdalkader

kwagyeman, 2018-06-09:
Wait, this is the fix here:

https://github.com/openmv/micropython/compare/2c25689c585655ab0a9374232a16915d4199af78...800d9c5206305447ed1a84de21c9f8ccc1544c6e#diff-43ae47295952eba3fe02829adfbac041

jonnor, 2024-09-29:
Is this still relevant on recent MicroPython versions, 1.23 or later?

iabdalkader, 2024-09-29:
I don't think this is relevant anymore, you can close this issue.

closed 2024-09-29 by jonnor