micropython/micropython #3855: ESP32 - machine.reset() IllegalInstruction

vshymanskyy, 2018-06-11:
When calling `machine.reset()` on ESP32, I get:
```
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction)
. Exception was unhandled.
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction)
. Exception was unhandled.
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction)
. Exception was unhandled.
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction)
. Exception was unhandled.
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction)
. Exception was unhandled.
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction)
. Exception was unhandled.
...
```
The board indeed resets after this ^_^ but I thing the expected behavior is different!

dpgeorge, 2018-06-12:
The same issue was recently fixed in #3810.  Did you use the latest firmware?  Maybe the bug resurfaced...

vshymanskyy, 2018-06-12:
I'll reopen if it appears on the latest. Provably a duplucate

closed 2018-06-12 by vshymanskyy