micropython/micropython #3856: Micropython does not raise TypeError for b"abc" + "def", but perhaps should

dstromberg, 2018-06-12:
It appears Micropython catenates these different types, while CPython 3.x and Pypy3 do not.  See below for a quick comparison - micropython appears at the end of the list:

$ pythons --command 'print(b"abc" + "def")'
cmd output started 2018 Tue Jun 12 10:24:05 AM PDT
/usr/local/cpython-1.0/bin/python (1.0.1) bad
      File "<string>", line 1
        print(b"abc" + "def")
                   ^
    SyntaxError: invalid syntax
/usr/local/cpython-1.1/bin/python (1.1) bad
      File "<string>", line 1
        print(b"abc" + "def")
                   ^
    SyntaxError: invalid syntax
/usr/local/cpython-1.2/bin/python (1.2) bad
      File "<string>", line 1
        print(b"abc" + "def")
                   ^
    SyntaxError: invalid syntax
/usr/local/cpython-1.3/bin/python (1.3) bad
      File "<string>", line 1
        print(b"abc" + "def")
                   ^
    SyntaxError: invalid syntax
/usr/local/cpython-1.4/bin/python (1.4) bad
      File "<string>", line 1
        print(b"abc" + "def")
                   ^
    SyntaxError: invalid syntax
/usr/local/cpython-1.5/bin/python (1.5.2) bad
      File "<string>", line 1
        print(b"abc" + "def")
                   ^
    SyntaxError: invalid syntax
/usr/local/cpython-1.6/bin/python (1.6.1) bad
      File "<string>", line 1
        print(b"abc" + "def")
                   ^
    SyntaxError: invalid syntax
/usr/local/cpython-2.0/bin/python (2.0.1) bad
      File "<string>", line 1
        print(b"abc" + "def")
                   ^
    SyntaxError: invalid syntax
/usr/local/cpython-2.1/bin/python (2.1.0) bad
      File "<string>", line 1
        print(b"abc" + "def")
                   ^
    SyntaxError: invalid syntax
/usr/local/cpython-2.2/bin/python (2.2.0) bad
      File "<string>", line 1
        print(b"abc" + "def")
                   ^
    SyntaxError: invalid syntax
/usr/local/cpython-2.3/bin/python (2.3.0) bad
      File "<string>", line 1
        print(b"abc" + "def")
                   ^
    SyntaxError: invalid syntax
/usr/local/cpython-2.4/bin/python (2.4.0) bad
      File "<string>", line 1
        print(b"abc" + "def")
                   ^
    SyntaxError: invalid syntax
/usr/local/cpython-2.5/bin/python (2.5.6) bad
      File "<string>", line 1
        print(b"abc" + "def")
                   ^
    SyntaxError: invalid syntax
/usr/local/cpython-2.6/bin/python (2.6.9) good abcdef
/usr/local/cpython-2.7/bin/python (2.7.13) good abcdef
/usr/local/cpython-3.0/bin/python (3.0.1) bad
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
    TypeError: can't concat bytes to str
/usr/local/cpython-3.1/bin/python (3.1.5) bad
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
    TypeError: can't concat bytes to str
/usr/local/cpython-3.2/bin/python (3.2.5) bad
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
    TypeError: can't concat bytes to str
/usr/local/cpython-3.3/bin/python (3.3.3) bad
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
    TypeError: can't concat bytes to str
/usr/local/cpython-3.4/bin/python (3.4.2) bad
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
    TypeError: can't concat bytes to str
/usr/local/cpython-3.5/bin/python (3.5.0) bad
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
    TypeError: can't concat bytes to str
/usr/local/cpython-3.6/bin/python (3.6.0) bad
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
    TypeError: can't concat bytes to str
/usr/local/cpython-3.7/bin/python (3.7.0b5) bad
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
    TypeError: can't concat str to bytes
/usr/local/jython-2.7/bin/jython (2.7.0) good abcdef
/usr/local/pypy-5.10.0/bin/pypy (2.7.13) good abcdef
/usr/local/pypy-5.3.1/bin/pypy (2.7.10) good abcdef
/usr/local/pypy-5.9.0/bin/pypy (2.7.13) good abcdef
/usr/local/pypy-6.0.0/bin/pypy (2.7.13) good abcdef
/usr/local/pypy3-5.10.0/bin/pypy3 (3.5.3) bad
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
    TypeError: unsupported operand type(s) for +: 'bytes' and 'str'
/usr/local/pypy3-5.5.0/bin/pypy3 (3.3.5) bad
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
    TypeError: unsupported operand type(s) for +: 'bytes' and 'str'
/usr/local/pypy3-5.8.0-with-lzma-fixes/bin/pypy3 (3.5.3) bad
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
    TypeError: unsupported operand type(s) for +: 'bytes' and 'str'
/usr/local/pypy3-5.8.0/bin/pypy3 (3.5.3) bad
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
    TypeError: unsupported operand type(s) for +: 'bytes' and 'str'
/usr/local/pypy3-5.9.0/bin/pypy3 (3.5.3) bad
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
    TypeError: unsupported operand type(s) for +: 'bytes' and 'str'
/usr/local/pypy3-6.0.0/bin/pypy3 (3.5.3) bad
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
    TypeError: unsupported operand type(s) for +: 'bytes' and 'str'
/usr/local/micropython-git-2017-06-16/bin/micropython (3.4.0) good b'abcdef'
/usr/local/micropython-git-2018-06-06/bin/micropython (3.4.0) good b'abcdef'

dpgeorge, 2018-06-13:
Thanks for the report, indeed MicroPython doesn't check this particular case.  This is because it's trying to be minimal.  In this case it's less restrictive than CPython (so doesn't take away functionality) and I don't really see a problem allowing this through.

dpgeorge, 2022-06-23:
I think it's OK to allow this, ie not raise an exception.  To be compatible with CPython just don't rely on this feature.

closed 2022-06-23 by dpgeorge

nmattia, 2025-08-04:
I think this also allows sockets to send `str`s: https://github.com/orgs/micropython/discussions/17834

Just to be clear, can the implicit conversion ever cause issues, or do both types have the exact same (C) representation?