micropython/micropython #3858: pyb.I2C doesn’t implement slave callback

adritium, 2018-06-13:
https://forum.micropython.org/viewtopic.php?t=4885

I am trying to use a STM32F board running MicroPython as a slave device.

There does seem to be some support for it in the pyb.I2C class but it doesn't seem to have a call back for when the master request.

dpgeorge, 2018-06-18:
> it doesn't seem to have a call back for when the master request.

That's correct, there is no callback.  It's not that easy to implement I2C slave, partly because uPy won't be fast enough to respond at a 400kHz SCL to each request for a byte.  It would be better to define a higher-level I2C slave interface (eg it just implements a memory-mapped register device like moste I2C devices out there) and then the callbacks happen much less frequently (eg end of entire transaction) than on a per-byte basis.

dpgeorge, 2022-06-23:
Closing due to inactivity.

closed 2022-06-23 by dpgeorge