micropython/micropython #3862: Decoding of base64 encoded file

Sunrise17, 2018-06-13:
@dpgeorge , I have a file encoded by "def encrypt()" in Python3 and i want to decode it in Micropython on ESP32 board, i have downloaded following [base64 module](https://github.com/micropython/micropython-lib/blob/master/base64/base64.py), and enabled urlsafe_b64decode() function (lines 136, 137, 138), and i have enabled lines 113 and 114 
    113: _urlsafe_encode_translation = bytes.maketrans(b'+/', b'-_')
    114: _urlsafe_decode_translation = bytes.maketrans(b'-_', b'+/')

Unfortunately, i am getting following error, It is not possible to decode it in MicroPython? or else, in what type of encryption method i can use it for MicroPython? I have already tried AES method but i could not find any library to decode it.

    File "base64.py", line 113, in <module>
    AttributeError: type object 'bytes' has no attribute 'maketrans'

    def encrypt(clear, key):
        enc = []
        for i in range(len(clear)):
            key_c = key[i % len(key)]
            enc_c = chr((clear[i] + key_c) % 256)
            enc.append(enc_c)
        return base64.urlsafe_b64encode("".join(enc).encode()).decode()

    def decrypt(enc, key):
        dec = []
        enc = base64.urlsafe_b64decode(enc).decode()
        for i in range(len(enc)):
            key_c = key[i % len(key)]
            dec_c = chr((256 + ord(enc[i]) - key_c) % 256)
            dec.append(dec_c)
        return "".join(dec)

peterhinch, 2018-06-14:
You should use libraries from [micropython-lib](https://github.com/micropython/micropython-lib.git) where you will find a base64 micro-implementation. In general these questions should be raised in [the forum](https://forum.micropython.org/) rather than in this bug tracker.

Sunrise17, 2018-06-14:
Dear @peterhinch , i have already used that library but when i enable the required lines as defined above, unfortunately, micropython does not support "bytes.maketrans", "translate" methods in due to its limitation i think. I will also open this issue in the forum.

aykevl, 2018-06-17:
Note: base64 is _not_ encryption. It is a way to encode binary data in ASCII-only streams, nothing more.
If you need encryption, use a library for that which abstracts away crypto primitives because using a primitive (like AES) directly [will only lead to problems](https://www.nccgroup.trust/us/about-us/newsroom-and-events/blog/2009/july/if-youre-typing-the-letters-a-e-s-into-your-code-youre-doing-it-wrong/).

dpgeorge, 2018-06-18:
The issues here seem to be:

1. Attempt to encrypt/decrypt data with uPy.  This is not yet supported but work has been done to get started in this direction, see #3460 
2. Attempt to use a modified micropython-lib/base64 library but it is not working, because str/bytes maketrans() and translate() methods are not supported by uPy.

As pointed out above, the `encrypt()` and `decrypt()` routines here are certainly not secure and shouldn't be used!  I suggest waiting until #3460 is merged and then use that to decrypt.

To address the particular issue of base64 not working, it's easy to just use `binascii.b2a_base64()` directly, eg:
```python
encoded = binascii.b2a_base64(data)
decoded = binascii.a2b_base64(encoded)
```

Sunrise17, 2018-06-18:
@dpgeorge , thanks for the response, i am looking forward to merged FW to use encryption and decryption. Following methods are working well to encode and decode the data. I think right now, it is not possible to add some simple encrypt/decrypt method on binascii.b2a_base64, right?

    encoded = binascii.b2a_base64(data)
    decoded = binascii.a2b_base64(encoded)

Sunrise17, 2018-06-18:
Ok, thanks for the information, looking forward to receive updated and merged FW consisting encrypt/decrypt methods.

dpgeorge, 2018-06-27:
ucryptolib is now available on unix, esp8266 and esp32.

closed 2018-06-27 by dpgeorge

Sunrise17, 2018-06-27:
Dear @dpgeorge , i have flashed the latest FW from git, how can i import 'ucryptolib' module to main function? In addition, i will be glad if you can let me know the basic use of the encrypt/decyrpt methods of ucryptolib. Thanks again for your efforts,

    ImportError: no module named 'ucryptolib'
    MicroPython v1.9.4-225-gd8dc918 on 2018-06-27; ESP32 module with ESP32

Sunrise17, 2018-06-27:
Dear @dpgeorge, Thanks for your effort, i have solved my issue with ucryptolib.

Sunrise17, 2018-06-27:
@dpgeorge , i have one question that in Python 3, to create random "IV is an initialization vector for CBC mode",  it is possible to do that with 'Crypto' module, but in MicroPython (as a simple form) which way you can advise to generate random IV vector?

    from Crypto import Random
    IV = Random.new().read(16)

dpgeorge, 2018-06-28:
> which way you can advise to generate random IV vector?

Try `IV = uos.urandom(16)`, that will work on CPython as well as uPy.

Please note that these kinds of questions are best asked on the forum: https://forum.micropython.org