micropython/micropython #3865: extmod/vfs.c:lookup_path() assumes string argument

peter-conalgo, 2018-06-13:
In the Unix port, I used to be able to 'open' an already-open file descriptor by number, like this:

```
f = open(45, 'rb')
```

but with recent changes for VFS support that doesn't work anymore, because the first argument to open is assumed to be a string by `lookup_path()`.

Maybe if an int is found here, we should assume it's the Posix filesystem? File descriptors are a uniquely Posix idea anyway.

dpgeorge, 2018-06-14:
The standard unix port should be working the same as before, so I assume you are using the coverage version?  Is there a reason to use this version over the standard, non-coverage version?

> Maybe if an int is found here, we should assume it's the Posix filesystem?

That's possible, and probably a good idea.

peter-conalgo, 2018-06-20:
I've build a simulator for my hardware using the unix port, so it's not one of the standard configurations for the Unix port. It would be "neat" to be mount a disk image as if it were a MicroSD card and operate at the FAT filesystem layer inside that. But also not critical, and yeah, I disabled the new VFS stuff as soon as I ran into this issue. Please consider this a low priority nit.

dpgeorge, 2018-07-03:
Fixed by a3c3dbd9559523937267fe5a24d86417e909ada7

closed 2018-07-03 by dpgeorge