micropython/micropython #3867: Ampy GET, PUT and LS commands do not work with esp8266

RobertoDebarba, 2018-06-13:
[Ampy](https://github.com/adafruit/ampy) GET, PUT and LS commands do not work with esp8266 via ampy.
I also opened a [issue on Ampy project](https://github.com/adafruit/ampy/issues/48).

Firmware: Micropython 1.9.4
Command: `sudo ampy --port /dev/ttyUSB0 --baud 115200 ls`

It works on Micropython 1.9.2. Same error with 1.9.3.

```
Traceback (most recent call last):
  File "/usr/local/bin/ampy", line 11, in <module>
    load_entry_point('adafruit-ampy==1.0.3', 'console_scripts', 'ampy')()
  File "/usr/local/lib/python3.6/dist-packages/click/core.py", line 722, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python3.6/dist-packages/click/core.py", line 697, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python3.6/dist-packages/click/core.py", line 1066, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python3.6/dist-packages/click/core.py", line 895, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python3.6/dist-packages/click/core.py", line 535, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python3.6/dist-packages/ampy/cli.py", line 147, in ls
    for f in board_files.ls(directory):
  File "/usr/local/lib/python3.6/dist-packages/ampy/files.py", line 103, in ls
    raise ex
  File "/usr/local/lib/python3.6/dist-packages/ampy/files.py", line 96, in ls
    out = self._pyboard.exec_(textwrap.dedent(command))
  File "/usr/local/lib/python3.6/dist-packages/ampy/pyboard.py", line 265, in exec_
    raise PyboardError('exception', ret, ret_err)
ampy.pyboard.PyboardError: ('exception', b'', b'Traceback (most recent call last):\r\n  File "<stdin>", line 6, in <module>\r\nAttributeError: no such attribute\r\n')
```

robert-hh, 2018-06-13:
No problem here. 
ampy, version 0.6.2
MicroPython v1.9.4-140-g8fb95d65-dirty on 2018-06-10; ESP module with ESP8266

AlexDanault, 2018-06-13:
Everything works here too, Micropython 1.94 on an ESP8266 (NodeMCU)

Did you erase_flash before flashing the firmware ?

RobertoDebarba, 2018-06-14:
I downloaded micropython 1.9.4 again, erase flash and write. And now ... It works!
I dont now if I did something wrong on first time, but tried everything e nothing worked.

Anyway, thanks for your support!

closed 2018-06-14 by RobertoDebarba

sany3001, 2023-12-28:
I have similar problem with MicroPython from ESP8266_GENERIC-20231005-v1.21.0.bin.
I cannot upload files using

`ampy --port /dev/ttyCH341USB0 put boot.py`
or
`ampy --port /dev/ttyCH341USB0 put boot.py /boot.py`

Always the same exception
`OSError: [Errno 19] ENODEV`

I can use ampy command `run` or `ls` but I cannot use `put`.

Linux Mint environment.