micropython/micropython #3869: NUCLEO_F091RC Timer pin output will not work

rolandvs, 2018-06-13:
Try to get a signal out of the STM32F091RC with

```
import pyb
ne555 = pyb.Timer(14)
signal_out = ne555.channel(2, mode=pyb.Timer.OC_TOGGLE, pin=pyb.Pin(pyb.Pin.cpu.A4))
```

This should work, however not only this but all things related to a pin, timer, callback and the above will output any signal.
Am I overlooking something or is some register not set correctly?

rolandvs, 2018-06-14:
With the STM32F091RC there is a timer `TIM1` that has some nice features like dead time insertion when using PWM and a break input to shut down the PWM generation in case of an event (error).

1) Will there be support for this timer, available in other STM32 derivates as `TIM1` and `TIM8` in the NEAR future?

In order to configure `TIM1` writing the registers directly might be an intermediate solution, however the `stm` module is disabled.

2) Is there a specific reason to disable the module `stm` in this port other than memory use?

dpgeorge, 2018-06-15:
I see that the original issue reported here (that timers don't work with pin output) has been deleted.  Please don't do that because it makes it hard to track the issues.

dpgeorge, 2018-06-15:
> Will there be support for this timer, available in other STM32 derivates as TIM1

Good that you noticed this, the omission of TIM1 was an oversight.  It is now fixed by 37a7257affa5e1ad449d911797dfc4b9e46677b3

> Is there a specific reason to disable the module stm in this port other than memory use?

It's only to reduce code size.  The stm module works fine on this board and you can enable it yourself if you like (see `MICROPY_PY_STM`).

rolandvs, 2018-06-17:
@dpgeorge I have a **feature** request.

I'm playing with the `TIM1` PWM with complimentary output. I use the dead-time in order to drive a half bridge. 

```
pin_CH1   = pyb.Pin(pyb.Pin.cpu.A8, mode=pyb.Pin.AF_PP, af=pyb.Pin.AF2_TIM1)
pin_CH1N = pyb.Pin(pyb.Pin.cpu.A7, mode=pyb.Pin.AF_PP, af=pyb.Pin.AF2_TIM1)
pin_BKIN  = pyb.Pin(pyb.Pin.cpu.A6, mode=pyb.Pin.AF_PP, af=pyb.Pin.AF2_TIM1)
timer_1  = pyb.Timer(1, freq=2000, mode=pyb.Timer.CENTER, deadtime=0)
timer1_ch1 = timer_1.channel(1, pyb.Timer.PWM, pulse_width=(timer_1.period()+1)//2, pin=pyb.Pin.cpu.A8)
```

A nice feature of `TIM1` is the `TIM1_BKIN` input to kill the PWM when something is wrong. The dead time setup in `timer.c` takes a few parameters like enabling the BKIN and invert polarity. Now I made quick fix by changing the assigned values, however it would be nice if the dead time configuration can be extended and also be set not only at the initialisation of the timer.

See `timer.c` `line 431...`

dpgeorge, 2018-06-18:
> I have a feature request.

Thanks for the idea.  But please don't overload an issue with more than one particular item because it gets hard to track the particular issue and to know when it is solved.

I'll close this issue because TIM1 is now supported on the F091.  I opened a new issue #3875 for the dead-time feature request.

closed 2018-06-18 by dpgeorge