micropython/micropython #3870: STM32: Using MICROPY_HW_FLASH_FS_LABEL

mcauser, 2018-06-14:
I tried adding a custom flash label to my `mpconfigboard.h`.
`#define MICROPY_HW_FLASH_FS_LABEL   "VET6MINI"`
Then put the board in DFU mode and flashed the latest MicroPython.

The previous `PYBFLASH` volume remains, including old files which I though would be overwritten.
It looks like it's not erasing the /flash volume on make deploy.
There's no `erase` task in the stm32 Makefile. Am I missing a step?

* Set BOOT0 to ON
* Connect USB
* make BOARD=VCC_GND_F407 deploy
* Disconnect USB
* Set BOOT0 to OFF
* Reconnect USB

```
$ make BOARD=VCC_GND_F407 deploy
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
Writing build-VCC_GND_F407/firmware.dfu to the board
File: build-VCC_GND_F407/firmware.dfu
    b'DfuSe' v1, image size: 324545, targets: 1
    b'Target' 0, alt setting: 0, name: "ST...", size: 324260, elements: 2
      0, address: 0x08000000, size: 14808
      1, address: 0x08020000, size: 309436
    usb: 0483:df11, device: 0x0000, dfu: 0x011a, b'UFD', 16, 0x7090dd56
Writing memory...
0x08000000   14808 [=========================] 100% 
0x08020000  309436 [=========================] 100% 
Exiting DFU...
```

mcauser, 2018-06-14:
Existing partitions:
```
$ dfu-util -l
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Deducing device DFU version from functional descriptor length
Found Runtime: [05ac:8289] ver=0137, devnum=6, cfg=1, intf=3, path="20-8.3", alt=0, name="UNKNOWN", serial="UNKNOWN"
Found DFU: [0483:df11] ver=2200, devnum=28, cfg=1, intf=0, path="20-5", alt=3, name="@Device Feature/0xFFFF0000/01*004 e", serial="396B35653436"
Found DFU: [0483:df11] ver=2200, devnum=28, cfg=1, intf=0, path="20-5", alt=2, name="@OTP Memory /0x1FFF7800/01*512 e,01*016 e", serial="396B35653436"
Found DFU: [0483:df11] ver=2200, devnum=28, cfg=1, intf=0, path="20-5", alt=1, name="@Option Bytes  /0x1FFFC000/01*016 e", serial="396B35653436"
Found DFU: [0483:df11] ver=2200, devnum=28, cfg=1, intf=0, path="20-5", alt=0, name="@Internal Flash  /0x08000000/04*016Kg,01*064Kg,07*128Kg", serial="396B35653436"
```

dpgeorge, 2018-06-14:
> It wasn't clear on the docs that flashing in DFU mode keeps the old filesystem.

It is definitely a feature that the old filesystem is kept!  IMO that's really important behaviour to retain.

> What do you think about adding an erase task to the stm32 Makefile to automate this?

You could try a factory reset of the board, that would recreate the filesystem and change the label.  That's anyway what happens after you do a full erase.

mcauser, 2018-06-14:
My confusion aside, I agree. The default behaviour should be to keep the existing filesystem.

I just think there should also be a full erase command added to the stm32 Makefile, like there is on the ESP8266 port.

rolandvs, 2018-06-14:
Full_erase would be a nice make option; in some boards I don't use any button, not able to reset the flash file system while booting as on the PYB boards.

**With regards to the name of the issue**: should we include the `MICROPY_HW_FLASH_FS_LABEL` in all `mpconfigboard.h` files that support an internal or SPI /flash disk?

dpgeorge, 2022-06-23:
Closing due to inactivity.

closed 2022-06-23 by dpgeorge